pibooth/pibooth #251: Move docs to readthedocs

sravel, 2021-05-27:
Readme is very big and if you want I can work to build a readthedocs website like this using sphinx:

[image]

doc is automatic update when git commit.

anxuae, 2021-05-28:
Hi @sravel , 

this is a good idea. I think you have almost do the job, can you propose a PR?
I have setup the Readthedoc project: https://pibooth.readthedocs.io/en/latest/

All documentation should be kept in the `docs/` folder.
We could reduce the `README.rst` to the strict minimum and by adding a link to the documentation

sravel, 2021-05-31:
[image]

anxuae, 2021-07-08:
Merge done

closed 2021-07-08 by anxuae