pibooth/pibooth #253: Temporary turn off "all" or "some" custom plugins from the pibooth.cfg with # problem

DJ-Dingo, 2021-05-28:
First, i love the turn on/off plugins in the menu.

But i try using # to disable all or some custom plugin scripts in the pibooth.cfg, and got and error
It is conflicting with the plugins_disabled = (´´)
I allso had plugins in the "plugins_disable" even that all plugins where turned on, maybe some of these where when i used # earlier in the custom plugin line (so i removed them manualy). 

What is the right way now,  to turn off the first or or some custom plugins with # in the config ,
You can only do it with plugins that comes after the first plugin, but not on the first custom plugin, and when you use # it disable ALL the rest of the plugins in the line. (I only use this when i test the difference between 2 scripts)

I try with ´´, # but that don´t work (It just says "Invalid Python module path" , i could offcouse make and empty plugin to be the first plugin, to make it work, but it still disable all the rest of my plugins.

As a workaround, i can make different plugin names on the different scripts, with the same plugin and then use plugins_disable or menu to control them. Just want to know what the right work flow is.

anxuae, 2021-05-28:
Hi @DJ-Dingo ,

The best is to give you some examples:

### One custom plugin installed

```ini
[GENERAL]
plugins = plugins/plugin.py
plugins_disabled = 
```

to disable the custom plugin:

```ini
[GENERAL]
plugins = plugins/my_plugin_hello.py
plugins_disabled = my_plugin_hello
```

**Note**: in the `plugins_disabled ` key you shall put the name of the disabled plugin (`my_plugin_hello`), not the path

### One custom plugin installed and official plugins

```ini
[GENERAL]
plugins = plugins/plugin.py
plugins_disabled = 
```

**Note**: the official plugins are not listed in the `plugins` key, because this key is only used to declare custom plugins (i.e. the ones not installed with `pip3`)

to disable the custom plugin:

```ini
[GENERAL]
plugins = plugins/my_plugin_hello.py
plugins_disabled = my_plugin_hello
```

to disable the custom plugin and some official plugins:

```ini
[GENERAL]
plugins = plugins/my_plugin_hello.py
plugins_disabled = ('pibooth_google_photo', 'my_plugin_hello', 'pibooth_qrcode')
```

Do I answer to your questions?

DJ-Dingo, 2021-05-28:
Hi @anxuae 

Yes you do, and i did understand it right from the beginning, i was just used to # in custom plugins, and got some extra plugins in the "plugins_disabled" that did not make sence.

And my problem was that some of my test plugins, is named Etc. my_plugin_hello_TEST_1.py, my_plugin_hello_TEST_2.py
But since they use the same GPIO, it gave problems to add both at the same time, since they can´t be turned on both at the same time, and pibooth would not start.

I just deleted the line in custom plugins, and start over with 1 plugin at a time, and then turn it off til i have all TEST plugins installed, then turn the once on that i am working on. I can work with that. Thx :-)

closed 2021-05-28 by DJ-Dingo