pibooth/pibooth #255: Pibooth blocked on "Thank you" page

pmag-p, 2021-06-05:
Hello everyone,

First of all, thanks a lot for your application and your work.
I am a beginner in RPI and it helped me a lot in my project for my father's birthday this summer.

Though for some days now I an having an issue, maybe you'll have an idea what's going wrong.

I am working with the editable/customizable version because I wanted to make some modifications, essentially esthetic (change the angle and size of the arrows, leds light continuously instead of blinking etc.)
It worked perfectly well, even after having operated my changes (or I would have known that I did something wrong).
Then I began to work on backgrounds and layouts, and I think that is when I started to have some problems.

I configured for a "1 photo" and a "3 photos" choice. The 2 templates share the same background, but have their own layout.
For the "1 photo" choice, everything works fine.

For the "3 photos" choice however, I have some trouble.
At the beginning, I had the following problem every 5 or 6 time or so, but now it is every time I choose the "3 photos" option.

Everything is ok until I arrive at the "final picture" step. I can see the picture with its background, its 3 photos and its overlay, everything seems to be normal (except that it can be a bit long, but this is not the main problem).
After that, whichever action is chosen (press "print", press "forget" or wait until it goes back to the main menu), it takes a veeery long time before it goes to the next step, in which you see the "Thank you" message. What's more, when it finally arrives ont the "Thank you" page,  it blocks in it forever, and the only thing I can do is stop the process (I even let it for more than 1 hour on the "Thank you" page to see if it was just very long, but it did not work, it did not come back on the main menu on its own).

Here are the steps indicated in the terminal, I can't see anything that went wrong in there :

```
[ INFO    ] pibooth           : Show picture choice (nothing selected)
[ INFO    ] pibooth           : Show picture choice (3 captures selected)
[ INFO    ] pibooth           : Show preview before next capture
[ INFO    ] pibooth           : Take a capture
[ INFO    ] pibooth           : Show preview before next capture
[ INFO    ] pibooth           : Take a capture
[ INFO    ] pibooth           : Show preview before next capture
[ INFO    ] pibooth           : Take a capture
[ INFO    ] pibooth           : Saving raw captures
[ INFO    ] pibooth           : Creating the final picture
[ INFO    ] pibooth           : Parsing pictures template file: /home/pi/.config/pibooth/picture_template_JC.xml
[ INFO    ] pibooth           : Found template 'Page-1': 1 captures - 0 texts
[ INFO    ] pibooth           : Found template 'Page-2': 2 captures - 2 texts
[ INFO    ] pibooth           : Found template 'Page-3': 3 captures - 0 texts
[ INFO    ] pibooth           : Found template 'Page-5': 1 captures - 2 texts
[ INFO    ] pibooth           : Found template 'Page-6': 2 captures - 2 texts
[ INFO    ] pibooth           : Found template 'Page-7': 3 captures - 2 texts
[ INFO    ] pibooth           : Found template 'Page-8': 4 captures - 2 texts
[ INFO    ] pibooth           : Use TemplatePictureFactory to create background
[ INFO    ] pibooth           : Use TemplatePictureFactory to concatenate images
[ INFO    ] pibooth           : Use TemplatePictureFactory to assemble final image
[ INFO    ] pibooth           : Use TemplatePictureFactory to draw texts
[ INFO    ] pibooth           : Save image '/media/pi/USB DISK/2021-06-05-10-56-12_pibooth.jpg'
[ INFO    ] pibooth           : Asyncronously generate pictures for animation
[ INFO    ] pibooth           : Uploading into EXISTING photo album -- '60 ans JC'
[ INFO    ] pibooth           : Uploading photo -- '/media/pi/USB DISK/2021-06-05-10-56-12_pibooth.jpg'
[ INFO    ] pibooth           : Added '2021-06-05-10-56-12_pibooth.jpg' to library and album '60 ans JC' 
[ INFO    ] pibooth           : Display the final picture
pygame 2.0.1 (SDL 2.0.9, Python 3.7.3)
Hello from the pygame community. https://www.pygame.org/contribute.html
pygame 2.0.1 (SDL 2.0.9, Python 3.7.3)
pygame 2.0.1 (SDL 2.0.9, Python 3.7.3)
Hello from the pygame community. https://www.pygame.org/contribute.html
Hello from the pygame community. https://www.pygame.org/contribute.html
```
_#Here is the moment when I'm blocked on the "Thank you" page and I can't do anaything more except stop the process. Then I have :_
```
Processus arr\u00eat\u00e9
pi@raspberrypi:~ $ /usr/lib/python3.7/multiprocessing/semaphore_tracker.py:144: UserWarning: semaphore_tracker: There appear to be 4 leaked semaphores to clean up at shutdown
  len(cache))
```

Once or twice, I had an error when pressing the "forget" button. Instead of just blocking on the "Thank you" page, I have the following : 

```
[ INFO    ] pibooth           : Display the final picture
[ INFO    ] pibooth           : Moving the picture in the forget folder
[ ERROR   ] pibooth           : 
[ ERROR   ] pibooth           : 
Traceback (most recent call last):
  File "/home/pi/pibooth/pibooth/booth.py", line 377, in main_loop
    self._machine.process(events)
  File "/home/pi/pibooth/pibooth/states.py", line 64, in process
    self.set_state(new_state_name)
  File "/home/pi/pibooth/pibooth/states.py", line 91, in set_state
    hook(cfg=self.cfg, app=self.app, win=self.win)
  File "/usr/local/lib/python3.7/dist-packages/pluggy/hooks.py", line 286, in __call__
    return self._hookexec(self, self.get_hookimpls(), kwargs)
  File "/usr/local/lib/python3.7/dist-packages/pluggy/manager.py", line 93, in _hookexec
    return self._inner_hookexec(hook, methods, kwargs)
  File "/usr/local/lib/python3.7/dist-packages/pluggy/manager.py", line 337, in traced_hookexec
    return outcome.get_result()
  File "/usr/local/lib/python3.7/dist-packages/pluggy/callers.py", line 80, in get_result
    raise ex[1].with_traceback(ex[2])
  File "/usr/local/lib/python3.7/dist-packages/pluggy/callers.py", line 52, in from_call
    result = func()
  File "/usr/local/lib/python3.7/dist-packages/pluggy/manager.py", line 335, in <lambda>
    outcome = _Result.from_call(lambda: oldcall(hook, hook_impls, kwargs))
  File "/usr/local/lib/python3.7/dist-packages/pluggy/manager.py", line 87, in <lambda>
    firstresult=hook.spec.opts.get("firstresult") if hook.spec else False,
  File "/usr/local/lib/python3.7/dist-packages/pluggy/callers.py", line 208, in _multicall
    return outcome.get_result()
  File "/usr/local/lib/python3.7/dist-packages/pluggy/callers.py", line 80, in get_result
    raise ex[1].with_traceback(ex[2])
  File "/usr/local/lib/python3.7/dist-packages/pluggy/callers.py", line 187, in _multicall
    res = hook_impl.function(*args)
  File "/home/pi/pibooth/pibooth/plugins/picture_plugin.py", line 69, in state_failsafe_enter
    self._reset_vars(app)
  File "/home/pi/pibooth/pibooth/plugins/picture_plugin.py", line 26, in _reset_vars
    self.factory_pool.clear()
  File "/home/pi/pibooth/pibooth/pictures/pool.py", line 28, in clear
    res.get(5)
  File "/usr/lib/python3.7/multiprocessing/pool.py", line 653, in get
    raise TimeoutError
multiprocessing.context.TimeoutError
[ ERROR   ] pibooth           : system='Linux', node='raspberrypi', release='5.10.17-v7+', version='#1414 SMP Fri Apr 30 13:18:35 BST 2021', machine='armv7l', processor=''
 ***********************************************************************************
 *                     Oops! It seems that pibooth has crached                     *
 *     You can report an issue on https://github.com/pibooth/pibooth/issues/new    *
 *                       and post the file: /tmp/pibooth.log                       *
 ***********************************************************************************
```

I attached the pibooth.log I have in this case.
[pibooth.log](https://github.com/pibooth/pibooth/files/6602340/pibooth.log)

Do you have any idea what's wrong ?

Thank you very much in advance for your help.

**Environment**
 - OS Raspbian 10 Buster
 - Camera Rpi camera - I had to set the GPU memory on 256 to make it work properly with the pibooth
 - Pibooth version 2.0.3
 -  Plugins used : templates, QRcode

anxuae, 2021-06-05:
Hi @pmag-p ,

Your issue seems to linked to the generation of animated image, can you confirm that you set the configuration `[WINDOW][animate] = True` ?
If yes, you should not encounter the issue if you deactivate the animation.

In order to investigate, can you provide me your overlay?

pmag-p, 2021-06-06:
Hello,

Thank you for your help.

Yes, I had `[WINDOW] [animate] = True`
I changed it for False, and indeed the problem disappeared.

I tried with 3 different overlays, for each of them if `[WINDOW] [animate] = True`, it still blocks on the "Thank you" page. 
Here is one of the overlay I tested :
[image]