pibooth/pibooth #258: Skip one picture and take it again

detrist, 2021-06-21:
Hi There,

I would like to work on a feature which would provide a screen with the raw images before the processing of the final image and would let the user to skip one of them and retake it.  Not sure if whether this would be a new plugin or not but I have not found any similar.

Any suggestions would be appreciated (e.g. which hook should be used to amend the existing feature set).

BR, detrist

guigro, 2021-07-03:
Great idea !

werdeil, 2021-07-05:
Hi, thanks for the enhancement suggestion. Our current point of view is that it is easier to restart a set of pictures than starting to rick infinite loops on skipping pictures. We tried to make the whole cycle (start - picture - processing - end) as fast as possible to let the users restart easily if they want.

I don't think I have seen this kind of feature on professional Photo Booth either, I guess for the same reason.

But I keep the issue open if someone has an idea on how to implement it.

DJ-Dingo, 2021-07-05:
Agree with @werdeil. It will take up too much time, if you have many users standing in line.
But many professional Photo Booth do have this future (for 5 sec). Personally, I find it annoying, because people delete and delete and delete all the time, and the funny pictures disappears. So better to let them take 1 extra photo if it´s not good enough.

anxuae, 2021-07-07:
Hi,

To be honest, I prefer to lets the user make a new sequence instead of retry the capture.
Some features could be tricky to implement.
Anyway, this kind of feature require an new state (because it is an new screen with new user actions)

anxuae, 2022-08-29:
Definitely, will not be implemented

closed 2022-08-29 by anxuae