pibooth/pibooth #259: Preview and countdown not working

Tonakka, 2021-06-21:
**Describe the bug**
The preview and Countdown are not appearing when taking a picture.

However, a simple script like this:
```
import picamera
from time import sleep

camera = picamera.PiCamera()
camera.resolution = (1280, 720)
camera.framerate = 24
camera.start_preview()

while True:
    sleep(1)
```

Does show a preview

**Steps to reproduce the bug**
1. Use the application normally. It just does not show the countdown nor the preview of the image

**Logs**
```
pi@raspberrypi:~ $ cat .config/pibooth/pibooth.cfg 
[GENERAL]
# User interface language: 'de', 'dk', 'en', 'es', 'fr', 'hu', 'nl' or 'no'
language = en

# Path to save pictures (list of quoted paths accepted)
directory = ~/Pictures/pibooth

# Start pibooth at Raspberry Pi startup
autostart = True

# In debug mode, exceptions are not caught, logs are more verbose, pictures are cleared at startup
debug = True

# Path to custom plugin(s) not installed with pip (list of quoted paths accepted)
plugins = 

# Plugin names to be disabled after startup (list of quoted names accepted)
plugins_disabled = 

# Enable a virtual keyboard in the settings interface
vkeyboard = False

[WINDOW]
# The (width, height) of the display window or 'fullscreen'
size = fullscreen

# Background RGB color or image path
background = (0, 0, 0)

# Text RGB color
text_color = (255, 255, 255)

# Blinking background when a capture is taken
flash = True

# Animate the last taken picture by displaying captures one by one
animate = False

# How long is displayed the capture in seconds before switching to the next one
animate_delay = 0.2

# On 'finish' state: how long is displayed the final picture in seconds (0 if never shown)
finish_image_delay = 0

# On 'wait' state: how long is displayed the final picture in seconds before being hidden (-1 if never hidden)
[... 244 lines cut ...]
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'tIME' 480 7
[ DEBUG   ] PIL.PngImagePlugin: b'tIME' 480 7 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'tEXt' 499 25
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'IDAT' 536 4879
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'IHDR' 16 13
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'iCCP' 41 388
[ DEBUG   ] PIL.PngImagePlugin: iCCP profile name b'ICC profile'
[ DEBUG   ] PIL.PngImagePlugin: Compression method 0
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'bKGD' 441 6
[ DEBUG   ] PIL.PngImagePlugin: b'bKGD' 441 6 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'pHYs' 459 9
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'tIME' 480 7
[ DEBUG   ] PIL.PngImagePlugin: b'tIME' 480 7 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'tEXt' 499 25
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'IDAT' 536 8192
[ DEBUG   ] pibooth           : Activate state 'wait'
[ DEBUG   ] pibooth           : Add to buffer the image '1566175312'
pi@raspberrypi:~ $ 

```

**Environment**
 - OS: PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
 - Camera; Rpi camera
 - Pibooth version 2..0.3

AonH, 2021-06-27:
Same Problem on here...

**Environment:**
- OS: PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
- Camera; RpiCam v2.1
- Pibooth version 2..0.3
- Raspberry 3 Model B+

werdeil, 2021-06-28:
Hi @Tonakka I see "No translation defined for 'en/capture' key" in the logs, did you changed the translation for this keyword? I can't test right now but I think your issue could come from this.

anxuae, 2021-07-07:
Hi,

After a clean install, with same version (using a 7'' Touch screen)
I'm not able to reproduce this issue, thus can not investigate.
I will have a try on a HDMI screen

anxuae, 2021-12-17:
Not able to reproduce

anxuae, 2022-04-05:
Please, re-open issue if reproduced

closed 2022-04-05 by anxuae