pibooth/pibooth #260: Choose between overlays

guigro, 2021-07-03:
Hello !
I'm digging into this project as I really like it !
I'm not a super dev, I'm just making trial and errors on config files and try to understand how things work.

It would be super good to have a list of overlays and to be able to pick the one you want for your picture !
Would be a first screen after clicking on the button to take a photo (just after the layout option).

Thanks !

werdeil, 2021-07-05:
That's a good idea, to be honest I didn't used a lot of overlays so didn't found myself in a situation that I wanted to use more than one at the same session.

DJ-Dingo, 2021-07-05:
I Like the idea

A suggestion:
I think it would be cool to have and extra RPI with a screen where users could make this changes. That way others can still take photos in the meantime

Etc. Ftp to another RPI where users could stand in peace and select the image and what overlays they want, and then have it sent, either via email, FB, or using a QR code. . 

Most times when I use my Pibooth, it is for a short period of only 2.5 hours where something between 400-700 pictures are taken. So it should preferably be as fast and smooth using my Pibooth.

guigro, 2021-07-05:
In fact where I want to put the photobooth, people have different scenes regarding where they went in the place, so they could select the overlay to choose the one corresponding to the scene.
They will know before taking the picture where they went, but it can be chosen later as well.

anxuae, 2021-07-07:
Hi,

### For the use of an extra RPI

In that case it should be another software, not pibooth.
The script `pibooth-gen` can be used to regenerate picture with different overlay

### For the selection in the current workflow

It will require a new state
Then a new Window screen design

anxuae, 2022-04-05:
Small slide-show to be used to select the background: https://github.com/anxuae/pygame-imslider

anxuae, 2022-04-19:
Hi @uncle-buzz ,

Unfortunately not, this would require to introduce a new state in pibooth which is not developed for the moment. Any PR will be welcome.

anxuae, 2022-04-21:
Hi @uncle-buzz 

Not possible through a plugin, and difficult to explain. I've planned to do it but need a rework of the window. So need time, maybe for summer.

uncle-buzz, 2022-04-22:
Hi @anxuae

Since there are 2 questions in my previous demand, I'm not sure your answer covered both or not, so simple question : is it possible with a plugin to propose more than 2 options for the number of pictures to take, if yes, with the help of the slide-show? 

Independently from this point, does the slide-show be usable with the touchscreen configuration ?

anxuae, 2022-04-22:
Not it is not possible from a plugin, it is necessary to interact deeper in the `pibooth` code

anxuae, 2022-04-22:
Yes the line show will be usable with a touch screen