pibooth/pibooth #261: Display preview for positioning before countdown

guigro, 2021-07-03:
Could be great to have another screen before countdown to get positioning.
It's nice when you take group photos to have this option that you can check how everybody fits in the frame, and then take the picture with or without countdown

werdeil, 2021-07-05:
Hi @guigro, don't you think that having a bigger countdown is enough? During this time most of the screen is already taken by the preview, I don't think adding a small text would change.

guigro, 2021-07-05:
Hello @werdeil ! A bigger countdown could work, but a preview screen may let the users place themself. Where I want to put the photobooth, the space is quite large and they don't know before making the setup where the limits are. And sometimes we shoot bigger groups (from 2 to 12 people).
I'll see if I can manage to make that through a plugin, that way it will be optional and not included into the main soft :)

closed 2021-07-05 by guigro

DJ-Dingo, 2021-07-05:
@guigro I use Gaffatape on the flor for that. 
A square that one must stay within. and then 2 x stripes in different colors. 1 for a close up´s(1-3 persons) and 1 further away for large groups. I use mine with kids and they learn pretty fast how it works for best pictures.