pibooth/pibooth #262: Display more than simple text in footer

guigro, 2021-07-05:
While putting my comments yesterday I forgot one !
I will try it on my own, but it would be nice to be able to put like the date or a variable text in the footer text.

anxuae, 2021-07-07:
Hi, 

Why do not put the date directly in the configuration file?
Using a `date` variable may be difficult because each country/language may have different syntax to display the date and thus to be portable/usable for all user, formatting could be not obvious.

But if you have severable example of variables, which make the feature really necessary, we could think about a way to implement it.

anxuae, 2021-07-09:
Can suppose:
- date
- count (picture number)
- user (raspberry-pi username)

anxuae, 2021-07-20:
See: https://pibooth.readthedocs.io/en/latest/sources/start.html#texts-and-fonts available in next `pibooth` version

commit d34a5df9 (2021-07-20): Fix #262: add 'date' and 'count' voraibles in footer texts
commit 93529b47 (2021-07-20): Fix #262: add 'date' and 'count' variables in footer texts