pibooth/pibooth #264: Command line camera control

sdw427, 2021-07-15:
I'm having trouble controlling my Fuji X-T2 with pibooth. As others have listed, the preview and the first image capture works perfectly, but then we get 'Oops something went wrong' as the camera hangs.

I have found with Reuterbal's photobooth program (also here on Github) - I can get full control of the camera when using their 'Gphoto2 command line' camera module. This simply runs the `--capture-image-and-download` command which seems to work perfectly. Their photobooth software however doesn't feature image preview when using this mode. 

As I'd like to make use of the various differences and enhancements (choice of images, google photo, and picture templates) in Pibooth, Is it possible to implement this kind of camera control, rather than using `GP_CAPTURE_IMAGE`?

Using this method, it relies on setting all the camera settings manually and not being able to change them , though with a flash, this should not be a problem in use.

Thanks!

anxuae, 2021-07-15:
Hi @sdw427 

Pibooth as been designed to take a sequence of capture then download all in one time. It permits to have a quick sequence of capture and performing long tasks (like downloading captures) during the "processing" state. If your camera does not support it, it could not be used with pibooth.

However more information are required in order to investigate:

1. Can you tell me if you can take picture with only one capture without error?

2. Can you paste the generated file `/tmp/pibooth.log` to ease investigation?

3. Can you run the tool `pibooth-diag` then paste here the generated file?

sdw427, 2021-07-23:
Thanks for the quick response @anxuae 

1. Yes, the first time I boot into Pibooth I am able to successfully preview and take 1 photo. Thereafter I am not able to do anything, I get 'oops something went wrong' message.

2. Results
[pibooth.log](https://github.com/pibooth/pibooth/files/6866088/pibooth.log)

3a. This is the result of the `pibooth-diag` immediately after my camera has crashed. 
[diagnostic.log](https://github.com/pibooth/pibooth/files/6866093/diagnostic.log)

3b And this is the results of the same test after reconnecting the camera.
[diagnostic.log](https://github.com/pibooth/pibooth/files/6866097/diagnostic.log)

Additional camera info / settings
- Raspberry Pi 4
- Camera connected to Pi with a USB 3 cable (2x USBA connectors into the PI, 1x USB micro-B into camera)
- ghptoto2 version - 2.5.27
- libgphoto2 version - 2.5.22
- libgphoto2_port - 0.12.0
- USB Tether Mode - AUTO
- Manual focus
- Auto ISO (camera minimum is 200)
- Auto shutter speed

Thanks again, any help would be fantastic.

anxuae, 2021-07-23:
Using command line, `pibooth` is using the following command line 4 times (in case of 4 sequence selected)
```
gphoto2 --capture-image
```
Then `pibooth` download the 4 captures.
This issue seems identical to #184 : Fuji Fujifilm X-T2 doesn't not support keeping several captures in memory before downloading them.

Do you have an option on your camera to select where the captures are saved ? For instance SDRAM, Memory card... ?

sdw427, 2021-07-26:
@anxuae 
I ran various iterations of this with no issues. I was able to run as many times as I want without crashing. 

#!/bin/bash

gphoto2 --capture-image
gphoto2 --capture-image
gphoto2 --capture-image
gphoto2 --capture-image
gphoto2 --capture-image
gphoto2 --capture-image

As soon as I introduce capture preview, things fail.

#!/bin/bash

gphoto2 --capture-image
gphoto2 --capture-preview
gphoto2 --capture-image
gphoto2 --capture-preview

Here is the error it spits out:
[Capturepreview error.txt](https://github.com/pibooth/pibooth/files/6881692/Capturepreview.error.txt)
Here is the result of the `gphoto2 --debug`:
[gphoto debug.txt](https://github.com/pibooth/pibooth/files/6881725/gphoto.debug.txt)

Digging into it a bit more, I can run `gphoto2 --capture-preview` indefinitely without failing however as soon as I run  `gphoto2 --capture-image`, I cannot run `gphoto2 --capture-preview` again without an error message. The camera indefinitely flashes the status LED (reg/green..) . Unplugging and resetting the camera fixes this.

I do not see an option anywhere to select where to store an image on my camera. 

Thanks.

anxuae, 2021-07-29:
Hi @sdw427 , maybe you can post your scripts in an issue of the [Gphoto2](https://github.com/gphoto/gphoto2) project and explains the error found in the **Capturepreview error.txt** file.

blutack, 2024-04-28:
I also have an X-T2 and ran into this issue. I've managed to fix it, although I don't want to open a pull request as the way I did it might cause issues with other cameras/platforms.

When the X-T2 takes a photo in tether mode, it always keeps a copy in the buffer (/store_1000001 or 2) and returns this path to gphoto. There is a separate setting that controls whether the camera additionally stores an image straight to the card (thanks to [jdalrym2](https://github.com/jdalrym2) commenting on https://github.com/gphoto/gphoto2/issues/443).

I want to keep the JPEGs on the card, so we can set this by doing `gphoto2 --set-config /main/other/d20c=3`. Note that the paths we get from gphoto2 will still be to the in memory buffers. There's no reason Pibooth couldn't set this and I'll open a separate pull request to do this.

From my testing that the X-T2 will not go into preview mode if there is a still an image in the memory buffer (/store_100000x) - it throws an error instead. This isn't a size issue, it even happens for tiny images.

In multi capture mode even with `delete_internal_memory = True`, Pibooth takes an image, leaves it in the camera buffer, switches to preview, takes the next image etc. After 4 images it then loops through downloading and deleting each image from the camera (if delete_internal_memory = True is set).

This is why the problem shows up in multicapture but not single capture mode - the X-T2 doesn't support switching to preview if there's an image in the buffer, and Pibooth doesn't delete images from the buffer in multicapture mode until all images have been taken.

The fix was pretty simple, I moved the download, open & delete from _post_process_capture into capture in gphoto.py. The X-T2 is so fast I don't see any visible difference but this obviously will have the effect of Pibooth using more memory & taking longer between shots (as we're keeping the full size PIL images in the tuples in `self._captures`). I'm running it on a Linux laptop so it doesn't bother me. I'm not sure if this is worth implementing for other cameras - it might be, I wouldn't be surprised if the Fuji ones aren't the only one with this limitation, and this would help for cameras with small internal memories, but there's probably a reason it's done this way (probably to mask the time it takes for slower DSLRs to transfer the images?).

For my patch below to work you must have delete_internal_memory = True set in your configuration file, don't worry - this doesn't remove the image from the SD card (remember, gphoto doesn't even know about that one), it instead removes it from the internal buffer.

I've tested this and can take 10 4-set captures in a row with no issues. If you want, I'm happy to open a pull request with this new behaviour gated behind a configuration option? That way, Pibooth would be usable with Fuji cameras (and cameras with small internal buffers) out of the box.

```python
    def capture(self, effect=None):
        """Capture a new picture.
        """
        if self._preview_viewfinder:
            self.set_config_value('actions', 'viewfinder', 0)

        effect = str(effect).lower()
        if effect not in self.IMAGE_EFFECTS:
            raise ValueError("Invalid capture effect '{}' (choose among {})".format(effect, self.IMAGE_EFFECTS))

        if self.capture_iso != self.preview_iso:
            self.set_config_value('imgsettings', 'iso', self.capture_iso)

        gp_image = self._cam.capture(gp.GP_CAPTURE_IMAGE)

        time.sleep(0.3)  # Necessary to let the time for the camera to save the image

        # used to be done in _post_process
        camera_file = self._cam.file_get(gp_image.folder, gp_image.name, gp.GP_FILE_TYPE_NORMAL)

        # used to be done in _post_process
        image = Image.open(io.BytesIO(camera_file.get_data_and_size()))

        self._captures.append((image, effect))

        if self.delete_internal_memory:
            LOGGER.debug("Delete capture '%s' from internal memory %s", gp_image.name, gp_image.folder)
            self._cam.file_delete(gp_image.folder, gp_image.name)

        if self.capture_iso != self.preview_iso:
            self.set_config_value('imgsettings', 'iso', self.preview_iso)

        self._hide_overlay()  # If stop_preview() has not been called

    def _post_process_capture(self, capture_data):
        """Rework capture data.

        :param capture_data: couple (GPhotoPath, effect)
        :type capture_data: tuple
        """
        # already an open PIL image at this point
        image, effect = capture_data
        # camera_file = self._cam.file_get(gp_path.folder, gp_path.name, gp.GP_FILE_TYPE_NORMAL)

        # if self.delete_internal_memory:
        #     LOGGER.debug("Delete capture '%s' from internal memory %s", gp_path.name, gp_path.folder)
        #     self._cam.file_delete(gp_path.folder, gp_path.name)

        #image = Image.open(io.BytesIO(camera_file.get_data_and_size()))

        image = self._rotate_image(image, self.capture_rotation)

        <function continues>
```