pibooth/pibooth #265: Support displaying "Get Ready" screen, even with only one capture format

mattdsteele, 2021-07-18:
**Is your feature request related to a problem? Please describe.**
Currently my configuration is setup for a single "choice" of capture numbers:

```
[PICTURE]
# Possible choice(s) of captures numbers (numbers between 1 to 4)
captures = (4)
```

When this configuration is set, it immediately moves to the preview countdown. This gives users a very limited amount of time to move away from the booth and prepare to be captured.

**Describe the solution you'd like**
I'd like for a new configuration option, perhaps something like `display_get_ready`, which when set to `True` would display the `ChosenScreen` for a set amount of time, before moving to the countdown.

**Describe alternatives you've considered**
Currently my "workaround" is to setup two "choices", by configuring `captures = (4, 4)`. This does display the `ChosenScreen` for a while, but it's definitely more confusing for users than it probably could be.

anxuae, 2021-07-20:
Hi @mattdsteele ,

Thanks for proposing enhancement. It is not currently foreseen to show the `ChosenScreen` when there is only one layout. But we can think about it.

However, a better workaround is to use the plugin described in the documentation in order to display a "Get Ready" message. See: https://pibooth.readthedocs.io/en/latest/sources/plugins/examples.html#add-get-ready-text-before-captures-sequence

mattdsteele, 2021-07-21:
That sounds great @anxuae, I was considering a plugin, but had no idea this was already documented!

anxuae, 2021-07-23:
@werdeil what do you think about always displaying the "Chosen" state even there is no choice on the number of captures?
It that case, increasing the countdown delay is not welcome, because it will increase the time before each capture. But the objective here is increasing the delay only before the first capture. Displaying the "Chosen" state is a way to do it.

commit 35182687 (2021-08-06): Fix #265: Chosen state always displayed (#275)

* Fix #265: Chosen state always displayed

* Add 'chosen_delay' option to set delay of chosen state

* Fix transient chosen state when delay is 0
closed 2021-08-06 by anxuae