pibooth/pibooth #267: app does not reinitialize camera if camera was disconnected

sandrokeil, 2021-07-24:
Thank you very much for this awesome project, it works like a charm. But there is one problem.

**Describe the bug**
The camera is not reinitialized after it was disconnected or was in standby mode.

**Steps to reproduce the bug**
1. Connect a DSLR camera and switch it on
2. Start pibooth
3. Take a photo
4. Disconnect camera
5. Take a photo
6. See error

Even after reconnecting camera the error still exists. Is it possible to reinitialize the camera so the app will work properly again? Another solution is to let the app die so it can be automatically restarted by systemd or supervisord.

**Logs**
```
pi@raspberrypi:~$ gphoto2 -v --auto-detect
gphoto2 2.5.27

Copyright (c) 2000-2021 Marcus Meissner and others

gphoto2 comes with NO WARRANTY, to the extent permitted by law. You may
redistribute copies of gphoto2 under the terms of the GNU General Public
License. For more information about these matters, see the files named COPYING.

This version of gphoto2 is using the following software versions and options:
gphoto2         2.5.27         gcc, popt(m), exif, no cdk, no aa, jpeg, no readline
libgphoto2      2.5.22         all camlibs, gcc, ltdl, EXIF
libgphoto2_port 0.12.0         iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi, gcc, ltdl, USB, serial without locking
Model                          Port                                            
----------------------------------------------------------
Canon EOS 70D                  usb:001,007     

pi@raspberrypi:~$ cat /tmp/pibooth.log 
[ INFO    ] pibooth           : Installed plugins: 
[ INFO    ] pibooth           : Starting the photo booth application on Raspberry pi 4B
[ DEBUG   ] pibooth           : Fallback to pibooth default camera management system
[ DEBUG   ] pibooth           : Found gPhoto2 cameras on ports: 'usb:001,006'
[ INFO    ] pibooth           : Configuring gPhoto2 camera ...
[ DEBUG   ] pibooth           : Getting option actions/viewfinder=2
[ DEBUG   ] pibooth           : Setting option imgsettings/iso=100
[ DEBUG   ] pibooth           : Setting option settings/capturetarget=Memory card
[ INFO    ] pibooth           : Connected to printer 'pibooth'
[ DEBUG   ] pibooth           : Activate state 'wait'
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'IHDR' 16 13
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'bKGD' 41 6
[ DEBUG   ] PIL.PngImagePlugin: b'bKGD' 41 6 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'pHYs' 59 9
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'tIME' 80 7
[ DEBUG   ] PIL.PngImagePlugin: b'tIME' 80 7 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'iTXt' 99 29
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'IDAT' 140 5756
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'IHDR' 16 13
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'zTXt' 41 750
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'iCCP' 803 388
[ DEBUG   ] PIL.PngImagePlugin: iCCP profile name b'ICC profile'
[ DEBUG   ] PIL.PngImagePlugin: Compression method 0
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'bKGD' 1203 6
[ DEBUG   ] PIL.PngImagePlugin: b'bKGD' 1203 6 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'pHYs' 1221 9
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'tIME' 1242 7
[ DEBUG   ] PIL.PngImagePlugin: b'tIME' 1242 7 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'IDAT' 1261 8192
[ DEBUG   ] pibooth           : Activate state 'choose'
[ INFO    ] pibooth           : Show picture choice (nothing selected)
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'IHDR' 16 13
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'iCCP' 41 388
[ DEBUG   ] PIL.PngImagePlugin: iCCP profile name b'ICC profile'
[ DEBUG   ] PIL.PngImagePlugin: Compression method 0
[... 354 lines cut ...]
  File "/usr/local/lib/python3.7/dist-packages/pluggy/manager.py", line 335, in <lambda>
    outcome = _Result.from_call(lambda: oldcall(hook, hook_impls, kwargs))
  File "/usr/local/lib/python3.7/dist-packages/pluggy/manager.py", line 87, in <lambda>
    firstresult=hook.spec.opts.get("firstresult") if hook.spec else False,
  File "/usr/local/lib/python3.7/dist-packages/pluggy/callers.py", line 208, in _multicall
    return outcome.get_result()
  File "/usr/local/lib/python3.7/dist-packages/pluggy/callers.py", line 80, in get_result
    raise ex[1].with_traceback(ex[2])
  File "/usr/local/lib/python3.7/dist-packages/pluggy/callers.py", line 187, in _multicall
    res = hook_impl.function(*args)
  File "/usr/local/lib/python3.7/dist-packages/pibooth/plugins/camera_plugin.py", line 68, in state_preview_enter
    app.camera.preview(win)
  File "/usr/local/lib/python3.7/dist-packages/pibooth/camera/gphoto.py", line 220, in preview
    self._window.show_image(self._get_preview_image())
  File "/usr/local/lib/python3.7/dist-packages/pibooth/camera/gphoto.py", line 129, in _get_preview_image
    cam_file = self._cam.capture_preview()
gphoto2.GPhoto2Error: [-52] Could not find the requested device on the USB port
[ DEBUG   ] pibooth           : Activate state 'failsafe'
[ ERROR   ] pibooth           : system='Linux', node='raspberrypi', release='5.10.17-v7l+', version='#1421 SMP Thu May 27 14:00:13 BST 2021', machine='armv7l', processor=''
```

**Environment**
 - OS Raspberry PI Debian 10 (Raspbian GNU/Linux 10 (buster))
 - Camera Canon EOS 70D
 - Pibooth version 2.0.3

sandrokeil, 2021-07-24:
A workaround is to enable the `debug` mode but there are red borders around the "buttons" then.

anxuae, 2021-07-25:
Hi @sandrokeil 

Thank you for reporting. The drivers of camera, OpenCV, GPhoto or RaspberryPi, are never re-initialized in case of hot-plug/unplug. Can you give me more details about why you observes disconnection of your DSLR?

A quick workaround to lets `pibooth` crash, could be to remove the `failsafe` state using a plugin:

```python
"""Plugin to remove the  'failsafe' state."""

import pibooth

__version__ = "1.0.0"

@pibooth.hookimpl
pibooth_startup(app)
    app._machine.remove_state('failsafe')
```

sandrokeil, 2021-07-25:
It was because the camera was configured to shutdown after some minutes to save power, but this can be deactivated. Thought it may be useful to reinitialize the camera in such a case if there is an error during taking a photo.

anxuae, 2022-04-05:
Deactivating the camera auto-shutdown solves the issue

closed 2022-04-05 by anxuae