pibooth/pibooth #268: Q: How to configure pibooth to print borderless with Canon Selphy CP1300?

sandrokeil, 2021-07-25:
The problem is that I have a small (white) border on the printed image but if I use `lp 2021-07-25-09-47-26_pibooth.jpg` the image is printed borderless like expected. Any help would be appreciated and this app is really cool.

How can I configure `pibooth` to use the default print settings of the system. The [documentation](https://pibooth.readthedocs.io/en/latest/sources/start.html?highlight=paper#printer) states "Here is the default configuration used for this project in CUPS, it may depend on the printer used" but I cannot find any configuration option to change it or to use the system printer settings.

Should an option like `{'fit-to-page':'True'}` added to the [`printFile`](https://github.com/pibooth/pibooth/blob/0096604b09d65f7cae3880d78a0bb14defff27d0/pibooth/printer.py#L106) definition? Best way would be to use the printer system settings or make it configurable to use system printer default settings.

### Sytem printer configuration

Here is how I configured the Canon Selphy CP1300 printer to print borderless for the postcard format (just for reference /cc @mattdsteele).

```
lpinfo -m | grep cp1300
lpadmin -p pibooth -E -v gutenprint53+usb://canon-cp1300/B721010709211348 -m gutenprint.5.3://canon-cp1300/expert
lpoptions -d pibooth
lpstat -s
lpstat -p pibooth
lpadmin -p pibooth -o printer-error-policy=abort-job

lpoptions -p pibooth -o StpBorderless=True
lpoptions -p pibooth -o StpiShrinkOutput=Expand
lpoptions -p pibooth -o StpColorPrecision=Best
lpoptions -p pibooth -o StpImageType=Photo

cupsenable pibooth
cupsaccept pibooth
```
The command `lpoptions -p pibooth -l` shows these default settings:

```
ageSize/Media Size: *Postcard w253h337 w155h244
ColorModel/Color Model: Gray Black *RGB CMY CMYK KCMY
StpColorPrecision/Color Precision: Normal *Best
Resolution/Resolution: *300dpi
StpiShrinkOutput/Shrink Page If Necessary to Fit Borders: Shrink Crop *Expand
StpBorderless/Borderless: False *True
StpColorCorrection/Color Correction: *None Accurate Bright Hue Uncorrected Desaturated Threshold Density Raw Predithered
StpBrightness/Brightness: 0 100 200 300 400 500 600 700 800 900 *None 1100 1200 1300 1400 1500 1600 1700 1800 1900 2000 Custom.REAL
StpFineBrightness/Brightness Fine Adjustment: *None 0 5 10 15 20 25 30 35 40 45 50 55 60 65 70 75 80 85 90 95
StpContrast/Contrast: 0 100 200 300 400 500 600 700 800 900 *None 1100 1200 1300 1400 1500 1600 1700 1800 1900 2000 2100 2200 2300 2400 2500 2600 2700 2800 2900 3000 3100 3200 3300 3400 3500 3600 3700 3800 3900 4000 Custom.REAL
StpFineContrast/Contrast Fine Adjustment: *None 0 5 10 15 20 25 30 35 40 45 50 55 60 65 70 75 80 85 90 95
StpSaturation/Saturation: 0 100 200 300 400 500 600 700 800 900 *None 1100 1200 1300 1400 1500 1600 1700 1800 1900 2000 2100 2200 2300 2400 2500 2600 2700 2800 2900 3000 3100 3200 3300 3400 3500 3600 3700 3800 3900 4000 4100 4200 4300 4400 4500 4600 4700 4800 4900 5000 5100 5200 5300 5400 5500 5600 5700 5800 5900 6000 6100 6200 6300 6400 6500 6600 6700 6800 6900 7000 7100 7200 7300 7400 7500 7600 7700 7800 7900 8000 8100 8200 8300 8400 8500 8600 8700 8800 8900 9000 Custom.REAL
StpFineSaturation/Saturation Fine Adjustment: *None 0 5 10 15 20 25 30 35 40 45 50 55 60 65 70 75 80 85 90 95
StpImageType/Image Type: None Text Graphics TextGraphics *Photo LineArt
StpGamma/Composite Gamma: 100 200 300 400 500 600 700 800 900 1000 1100 1200 1300 1400 1500 1600 1700 1800 1900 2000 2100 2200 2300 2400 2500 2600 2700 2800 2900 3000 3100 3200 3300 3400 3500 3600 3700 3800 3900 4000 *None Custom.REAL
StpFineGamma/Composite Gamma Fine Adjustment: *None 0 5 10 15 20 25 30 35 40 45 50 55 60 65 70 75 80 85 90 95
StpCyanGamma/Cyan: 0 100 200 300 400 500 600 700 800 900 1000 1100 1200 1300 1400 1500 1600 1700 1800 1900 2000 2100 2200 2300 2400 2500 2600 2700 2800 2900 3000 3100 3200 3300 3400 3500 3600 3700 3800 3900 4000 *None Custom.REAL
StpFineCyanGamma/Cyan Fine Adjustment: *None 0 5 10 15 20 25 30 35 40 45 50 55 60 65 70 75 80 85 90 95
StpMagentaGamma/Magenta: 0 100 200 300 400 500 600 700 800 900 1000 1100 1200 1300 1400 1500 1600 1700 1800 1900 2000 2100 2200 2300 2400 2500 2600 2700 2800 2900 3000 3100 3200 3300 3400 3500 3600 3700 3800 3900 4000 *None Custom.REAL
StpFineMagentaGamma/Magenta Fine Adjustment: *None 0 5 10 15 20 25 30 35 40 45 50 55 60 65 70 75 80 85 90 95
StpYellowGamma/Yellow: 0 100 200 300 400 500 600 700 800 900 1000 1100 1200 1300 1400 1500 1600 1700 1800 1900 2000 2100 2200 2300 2400 2500 2600 2700 2800 2900 3000 3100 3200 3300 3400 3500 3600 3700 3800 3900 4000 *None Custom.REAL
StpFineYellowGamma/Yellow Fine Adjustment: *None 0 5 10 15 20 25 30 35 40 45 50 55 60 65 70 75 80 85 90 95
StpCyanBalance/Cyan Balance: 0 100 200 300 400 500 600 700 800 900 1000 *None Custom.REAL
StpFineCyanBalance/Cyan Balance Fine Adjustment: *None 0 5 10 15 20 25 30 35 40 45 50 55 60 65 70 75 80 85 90 95
StpMagentaBalance/Magenta Balance: 0 100 200 300 400 500 600 700 800 900 1000 *None Custom.REAL
StpFineMagentaBalance/Magenta Balance Fine Adjustment: *None 0 5 10 15 20 25 30 35 40 45 50 55 60 65 70 75 80 85 90 95
StpYellowBalance/Yellow Balance: 0 100 200 300 400 500 600 700 800 900 1000 *None Custom.REAL
StpFineYellowBalance/Yellow Balance Fine Adjustment: *None 0 5 10 15 20 25 30 35 40 45 50 55 60 65 70 75 80 85 90 95
StpLinearContrast/Linear Contrast Adjustment: *False True
```

### Config
This is my `pibooth v2.0.3` configuration:

```
[GENERAL]
# User interface language: 'de', 'dk', 'en', 'es', 'fr', 'hu', 'nl' or 'no'
language = de 

# Path to save pictures (list of quoted paths accepted)
directory = ~/Pictures/pibooth
[... 124 lines cut ...]
# Maximum number of duplicate pages sent to the printer (avoid paper waste)
max_duplicates = 1 

# Print 1, 2, 3 or 4 picture copies per page
pictures_per_page = 1

[CONTROLS]
# How long to press a single hardware button in seconds
debounce_delay = 0.3

# How long to press multiple hardware buttons in seconds
multi_press_delay = 0.5

# Physical GPIO IN pin to take a picture
picture_btn_pin = 11

# Physical GPIO OUT pin to light a LED when picture button is pressed
picture_led_pin = 7

# Physical GPIO IN pin to print a picture
print_btn_pin = 13

# Physical GPIO OUT pin to light a LED when print button is pressed
print_led_pin = 15
```

sandrokeil, 2021-07-25:
I'm playing around to figure out why the system settings are not used and it looks like python cups does not use the system settings. It uses the common default printer settings. I added the following lines at the end of the [printer.py __init__](https://github.com/pibooth/pibooth/blob/master/pibooth/printer.py#L35) method.

```
self.printersettings = dict()
f = self._conn.getPPD(self.name)
ppd = cups.PPD(f)
#ppd.markDefaults()
groups = ppd.optionGroups
option2selection = dict()
for group in groups:
    for opt in group.options:
        selection = list(map(lambda x: x["text"], opt.choices))
        option2selection[opt] = selection
self.printersettings[self.name] = option2selection

LOGGER.info("printer settings: %s", str(self.printersettings))
```

Which results in the log output beneath. For instance `cups.Option StpBorderless=False` should be `True`, because the it's set to `true` via `lpoptions -p pibooth -o StpBorderless=True` before.

> [ INFO    ] pibooth           : printer settings: {'pibooth': {<cups.Option PageSize=Postcard>: ['Postcard 100x148mm', 'CP_L 89x119mm', 'Card 54x86mm'], <cups.Option PageRegion=Postcard>: ['Postcard 100x148mm', 'CP_L 89x119mm', 'Card 54x86mm'], <cups.Option ColorModel=RGB>: ['Grayscale', 'Inverted Grayscale', 'RGB Color', 'CMY Color', 'CMYK', 'KCMY'], <cups.Option StpColorPrecision=Normal>: ['Normal', 'Best'], <cups.Option Resolution=300dpi>: ['300x300 DPI'], <cups.Option StpiShrinkOutput=Shrink>: ['Shrink (print the whole page)', 'Crop (preserve dimensions)', 'Expand (use maximum page area)'], <cups.Option StpBorderless=False>: ['No', 'Yes'], <cups.Option StpColorCorrection=None>: ['Default', 'High Accuracy', 'Bright Colors', 'Correct Hue Only', 'Uncorrected', 'Desaturated', 'Threshold', 'Density', 'Raw', 'Pre-Dithered'], <cups.Option StpBrightness=None>: ['0.000', '0.100', '0.200', '0.300', '0.400', '0.500', '0.600', '0.700', '0.800', '0.900', '1.000', '1.100', '1.200', '1.300', '1.400', '1.500', '1.600', '1.700', '1.800', '1.900', '2.000', 'Custom'], <cups.Option StpFineBrightness=None>: ['0.000', '0.000', '0.005', '0.010', '0.015', '0.020', '0.025', '0.030', '0.035', '0.040', '0.045', '0.050', '0.055', '0.060', '0.065', '0.070', '0.075', '0.080', '0.085', '0.090', '0.095'], <cups.Option StpContrast=None>: ['0.000', '0.100', '0.200', '0.300', '0.400', '0.500', '0.600', '0.700', '0.800', '0.900', '1.000', '1.100', '1.200', '1.300', '1.400', '1.500', '1.600', '1.700', '1.800', '1.900', '2.000', '2.100', '2.200', '2.300', '2.400', '2.500', '2.600', '2.700', '2.800', '2.900', '3.000', '3.100', '3.200', '3.300', '3.400', '3.500', '3.600', '3.700', '3.800', '3.900', '4.000', 'Custom'], <cups.Option StpFineContrast=None>: ['0.000', '0.000', '0.005', '0.010', '0.015', '0.020', '0.025', '0.030', '0.035', '0.040', '0.045', '0.050', '0.055', '0.060', '0.065', '0.070', '0.075', '0.080', '0.085', '0.090', '0.095'], <cups.Option StpSaturation=None>: ['0.000', '0.100', '0.200', '0.300', '0.400', '0.500', '0.600', '0.700', '0.800', '0.900', '1.000', '1.100', '1.200', '1.300', '1.400', '1.500', '1.600', '1.700', '1.800', '1.900', '2.000', '2.100', '2.200', '2.300', '2.400', '2.500', '2.600', '2.700', '2.800', '2.900', '3.000', '3.100', '3.200', '3.300', '3.400', '3.500', '3.600', '3.700', '3.800', '3.900', '4.000', '4.100', '4.200', '4.300', '4.400', '4.500', '4.600', '4.700', '4.800', '4.900', '5.000', '5.100', '5.200', '5.300', '5.400', '5.500', '5.600', '5.700', '5.800', '5.900', '6.000', '6.100', '6.200', '6.300', '6.400', '6.500', '6.600', '6.700', '6.800', '6.900', '7.000', '7.100', '7.200', '7.300', '7.400', '7.500', '7.600', '7.700', '7.800', '7.900', '8.000', '8.100', '8.200', '8.300', '8.400', '8.500', '8.600', '8.700', '8.800', '8.900', '9.000', 'Custom'], <cups.Option StpFineSaturation=None>: ['0.000', '0.000', '0.005', '0.010', '0.015', '0.020', '0.025', '0.030', '0.035', '0.040', '0.045', '0.050', '0.055', '0.060', '0.065', '0.070', '0.075', '0.080', '0.085', '0.090', '0.095'], <cups.Option StpImageType=TextGraphics>: ['Manual Control', 'Text', 'Graphics', 'Mixed Text and Graphics', 'Photograph', 'Line Art'], <cups.Option StpGamma=None>: ['0.100', '0.200', '0.300', '0.400', '0.500', '0.600', '0.700', '0.800', '0.900', '1.000', '1.100', '1.200', '1.300', '1.400', '1.500', '1.600', '1.700', '1.800', '1.900', '2.000', '2.100', '2.200', '2.300', '2.400', '2.500', '2.600', '2.700', '2.800', '2.900', '3.000', '3.100', '3.200', '3.300', '3.400', '3.500', '3.600', '3.700', '3.800', '3.900', '4.000', 'None', 'Custom'], <cups.Option StpFineGamma=None>: ['0.000', '0.000', '0.005', '0.010', '0.015', '0.020', '0.025', '0.030', '0.035', '0.040', '0.045', '0.050', '0.055', '0.060', '0.065', '0.070', '0.075', '0.080', '0.085', '0.090', '0.095'], <cups.Option StpCyanGamma=None>: ['0.000', '0.100', '0.200', '0.300', '0.400', '0.500', '0.600', '0.700', '0.800', '0.900', '1.000', '1.100', '1.200', '1.300', '1.400', '1.500', '1.600', '1.700', '1.800', '1.900', '2.000', '2.100', '2.200', '2.300', '2.400', '2.500', '2.600', '2.700', '2.800', '2.900', '3.000', '3.100', '3.200', '3.300', '3.400', '3.500', '3.600', '3.700', '3.800', '3.900', '4.000', 'None', 'Custom'], <cups.Option StpFineCyanGamma=None>: ['0.000', '0.000', '0.005', '0.010', '0.015', '0.020', '0.025', '0.030', '0.035', '0.040', '0.045', '0.050', '0.055', '0.060', '0.065', '0.070', '0.075', '0.080', '0.085', '0.090', '0.095'], <cups.Option StpMagentaGamma=None>: ['0.000', '0.100', '0.200', '0.300', '0.400', '0.500', ' [... cut ...] 0.070', '0.075', '0.080', '0.085', '0.090', '0.095'], <cups.Option StpYellowBalance=None>: ['0.000', '0.100', '0.200', '0.300', '0.400', '0.500', '0.600', '0.700', '0.800', '0.900', '1.000', 'None', 'Custom'], <cups.Option StpFineYellowBalance=None>: ['0.000', '0.000', '0.005', '0.010', '0.015', '0.020', '0.025', '0.030', '0.035', '0.040', '0.045', '0.050', '0.055', '0.060', '0.065', '0.070', '0.075', '0.080', '0.085', '0.090', '0.095'], <cups.Option StpLinearContrast=False>: ['No', 'Yes']}}
```

sandrokeil, 2021-07-25:
As a workaround I just changed the print line to

```
self._conn.printFile(self.name, filename, osp.basename(filename), {'StpBorderless': 'True', 'StpiShrinkOutput': 'Expand', 'StpColorPrecision': 'Best', 'StpImageType': 'Photo'})        
```

What do you think about adding printer specific configuration to the pibooth config file e.g.

```
[PRINTER:pibooth]
StpBorderless = True
options.StpiShrinkOutput = Expand
options.StpColorPrecision = Best
options.StpImageType = Photo
```

anxuae, 2021-07-25:
Hi @sandrokeil,

I'm never configured the printer through lpadmin. Generally I'm using the CUPS interface available at http://localhost:631/printers .

But I could add an option to pass extra parameters to the printer.

Maybe you can also report an issue to the https://github.com/OpenPrinting/pycups project?

anxuae, 2021-07-25:
You can try the code in #270, I added a new configuration option `printer_options` where you can define a dictionary of CUPS options as defied in the PPD file.

I also add a new command `pibooth-printcfg` to list all available options

sandrokeil, 2021-07-25:
> I'm never configured the printer through lpadmin. Generally I'm using the CUPS interface available at http://localhost:631/printers 

Indeed, if I use the webinterface and change the default settings for the printer, they are now available in `pibooth`. Thank you very much for the printer config settings overview. :+1:  I can confirm that it works.

closed 2021-07-25 by sandrokeil

mattdsteele, 2021-07-25:
@sandrokeil @anxuae This is great, thank you for digging into this! I have a Selphy CP510 I can give this a shot with tomorrow as well.

To summarize the discussion and so confirm my understanding, I'll have to switch to the dev build (or wait for a release), and then add this section to my config:

```
[PRINTER]
printer_options = {'StpBorderless': 'True', 'StpImageType': 'Photo'}
```

And I shouldn't have to configure anything in CUPS or lpadmin?

sandrokeil, 2021-07-28:
@mattdsteele Yes, the printer options needs a new release or the dev build. You can try it with these two options, maybe it's enough. See [this comment](https://github.com/pibooth/pibooth/pull/270#issuecomment-886229283) to see my default printer settings which I changed via the webinterface. With these settings and the two options above I can print borderless pictures.

werdeil, 2021-07-29:
So just to avoid adding code that is not necessary you managed to make the printer work as expected without changes in the code or not?

sandrokeil, 2021-07-29:
@werdeil The `printer_option` option and the current printer setting script  from PR #270 is very helpful for debugging reasons and I don't have seen the options `StpBorderless` and `StpImageType` in the web interface. So the `printer_option` option is also needed for detailed configuration.

No other additional code is needed. I think you can create a new release.

mattdsteele, 2021-07-31:
I can confirm that updating the CUPS options via the web is insufficient to print borderless; the additional printer options from #270 is needed to get it working on my Selphy CP510. Thanks!

commit 7727a775 (2021-07-25): Fix #268: add printer_options configuration key (#270)

* Fix #268: add printer_options configuration key

* Add printer options in debug logs