pibooth/pibooth #271: Disambiguate touch areas

forthib, 2021-07-25:
Hi!

I tried the "arrows = touchscreen" setting. It works great, but some touch areas are ambiguous.
During the "print" state, the user has three possible actions: #1 print (and keep) the picture #2 delete the picture #3 do nothing for n seconds == keep the picture without printing. Action #1 is easy to activate: the zone to touch corresponds to the "printer" pictogram. Action #3 is OK. Action #2 is tricky: the user has to touch the left part of the screen to currently delete the picture. If he touches the text "Please forget this photo", it actually prints the picture.
[image]

Same behaviour in the "wait" state: if the user touch "Or you can still pring this photo", it's like he touched "Take a photo".
[image]

Is it possible to either:
 - change the behaviour: make touching these text areas cause the associated action
 - move these text areas to the other side of the screen, over the picture, and possibly replace text by pictogram (recycle bin, printer)
 - add a setting to remove these two actions : no remove action on the "print" state, no print action on the "wait" state.

werdeil, 2021-07-29:
Hi @forthib we indeed choose to have a simple behavior on touch to make it work with all screens : separate the screen in two half.

That's why we tried or best to indicate where to press with pictos. 

We thought about changing the text to something like "press the picture to ..." But as we use the same text for physical buttons it won't work. However if you are using a touchscreen only device you can put your own wording (through the translate.cfg file) to make it easier to understand.

Moving text over the image is not a good solution as depending on the layout we may hide the best part of the image. 

Disabling the features could be a solution but I'm not a fan of hiding features.

anxuae, 2023-03-11:
Fixed in 3.0.0 that will be release soon

closed 2023-03-11 by anxuae