pibooth/pibooth #273: Automatically print a photo, and not display the prompt

anxuae, 2021-08-04:
Printing question: is there a way to automatically print a photo, and not display the prompt?

I'd prefer not to wire up an additional button or UI for folks, and just have a single print be automatic; similar to the way "traditional" photo booths work.

_Originally posted by @mattdsteele in https://github.com/pibooth/pibooth/issues/42#issuecomment-875744758_

anxuae, 2021-08-04:
@mattdsteele I'ved opened an issue, because for the moment, there is no way to it, but it could be a useful improvement

mattdsteele, 2021-08-05:
+1 for this config option 🎉 

For my use case, I'd also like for this to work with `max_pages`, as we'd want to stop printing after the paper has been exhausted.

closed 2021-08-19 by anxuae

NiraCity, 2024-05-08:
These are my setting.
This is easier than described above

# Name of the printer defined in CUPS (or use the 'default' one)
printer_name = default

# Print options passed to the printer, shall be a valid Python dictionary
printer_options = {}

# How long is the print view in seconds (0 to skip it)
printer_delay = 10

# Number of pages automatically sent to the printer (or use 'max' to reach max duplicate)
auto_print = 1

# Maximum number of printed pages before warning on paper/ink levels (-1 = infinite)
max_pages = -1

# Maximum number of duplicate pages sent to the printer (avoid paper waste)
max_duplicates = 3

# Print 1, 2, 3 or 4 picture copies per page
pictures_per_page = 1

commit 3069d168 (2021-08-19): Fix #273: Add 'auto_print' option to automatically print a picture (#276)

* Fix #273: add 'auto_print' option to automatically print a picture

* Update option description

* Rename print() to print_picture(), add possible value auto_print=max