pibooth/pibooth #278: Capture video stream and use it for preview

fschrempf, 2021-08-15:
**Is your feature request related to a problem? Please describe.**
Some DSLRs don't support live view via gphoto2, but have a video port (HDMI, Composite, etc.) that can be used to get a live image for previewing.

**Describe the solution you'd like**
It would be nice if it would be possible to use a cheap USB HDMI capture adapter and include the stream from this device as preview image in pibooth. This is probably not that hard to implement as it seems like the "hybrid mode" already does something similar if I understand this correctly.

anxuae, 2021-08-16:
Hi @fschrempf , could be a good enhancement. Do you have any examples of implementation or library that can be used to achieve that?

fschrempf, 2021-08-16:
Ho @anxuae, no I haven't looked at the details yet and don't know how to implement this. This was just an idea that I wanted to share.

As far as I know a USB HDMI capture adapter would be exposed as video (v4l2) device in Linux. And the RPi camera is probably also exposed as video device. So interfacing with the adapter should be the same as for the RPi camera. At least that's what I would expect without looking at any code :wink:

fschrempf, 2021-08-21:
After having a closer look, I assume that this is already possible with the current hybrid camera feature using OpenCV. The HDMI grabber will act just like a webcam, so there's no need for any additional code to support this. The only thing that might be missing is a config option to select the OpenCV port if multiple devices are present (e. g. if you attach a HDMI grabber to laptop with builtin webcam).

closed 2021-08-21 by fschrempf

Yllesia, 2021-09-17:
Hi @fschrempf ,
Did you succeed to use your HDMI to USB converter to get the image directly from the DSLR camera ?

fschrempf, 2021-09-21:
> Did you succeed to use your HDMI to USB converter to get the image directly from the DSLR camera ?

I tried it and as noted above, the pibooth hybrid camera mode automatically detects the HDMI grabber as OpenCV device and uses it for previewing. The problem I experienced was with the camera (Pentax K-5 II) which didn't allow to control it with gphoto2 via USB **and at the same time** also output the preview image via HDMI. Whenever I start remote controlling the camera via USB, the LCD and HDMI image turns blank.

Binklebonk, 2021-12-15:
I can confirm similar behaviour with Nikon D750.  When USB is connected, LiveView on the camera is not possible.  HDMI output is dependent on LiveView being active.

Essentially, it's either USB or HDMI, not both.  It's not even possible to switch between them.  So long as the USB port is _connected_, HDMI output is not possible.