pibooth/pibooth #279: Problem with picamera

SergioMuinhos, 2021-08-30:
Hi, 
When i take a picute said that "woops i have a problem".
I will paste the log.
I checked all conetions and are all OK.
Please help me. Thanks
[pibooth.log](https://github.com/pibooth/pibooth/files/7077470/pibooth.log)

anxuae, 2021-08-30:
Hi @SergioMuinhos 

From the log:
```python
  File "/usr/lib/python3/dist-packages/picamera/camera.py", line 1995, in _control_callback
    "No data recevied from sensor. Check all connections, "
picamera.exc.PiCameraRuntimeError: No data recevied from sensor. Check all connections, including the SUNNY chip on the camera board
```

It seems that your camera is not correctly connected or you don't enable the camera in [raspi-config](https://www.raspberrypi.org/documentation/computers/configuration.html)

sergiomgarc, 2021-08-30:
Hi, @anxuae
On The first thanks for your answer.
The camera is corrected connected and enabled in raspi-config 
[image]
[image]
Could it be that the camera has broken down? 🤔 
Best regards

anxuae, 2021-08-30:
Googling the error, maybe you can try some suggestions : https://stackoverflow.com/questions/31354280/raspberry-camera-error-mmal-no-data-received-from-sensor

https://forum.core-electronics.com.au/t/camera-board-v2-issue/1357

DJ-Dingo, 2021-08-30:
Hi @sergiomgarc 
When i had this problem it was the cable connection. 
You can test your cam with this in terminal

Take a photo 
raspistill -o testshot.jpg

Make a 5 sec video
raspivid -o testvideo.h264 -t 5000

It should make a video "testvideo.h264 or photo "testshot.jpg" in the folder you are in.

DJ-Dingo, 2021-08-30:
@sergiomgarc 

After looking closer at your pics, it looks like the cable from the cam to the RPI is wrong, 
I think it should be turnt around on the raspberry pi.  It looks right on the cam, but i dont have taht one so can tell for sure.
So check you cable connections on the RPI first

SergioMuinhos, 2021-08-31:
Hi @DJ-Dingo 
All connections are correct. Otherwise, the Raspberry would not even put that error.
Looking at the links that were attached by @anxuae It's possible that the sensor died. And I will buy another sensor for test.
In brief I will put if it is a solution

DJ-Dingo, 2021-08-31:
@SergioMuinhos. 
OK. But mine did the same and it was the cable connection. And when I looked at your picture and lined up with my own, it looked wrong with the connection on the rpi. But you should know best, of course :-)

anxuae, 2021-09-22:
@SergioMuinhos did you try with a new camera?

sergiomgarc, 2021-09-22:
Yes, Just arrived today, and finally the old camera is died. The New camera works perfectly.

closed 2021-09-22 by anxuae