pibooth/pibooth #280: "Oops something wrong happens" when trying to take a photo on Fullscreen Pibooth

Yllesia, 2021-09-05:
Hi everybody, i'm trying to use Pibooth to make a photobooth for a wedding on 16th of October but i'm having some issues right now. It's very important to me to have something that work fine especially on this wedding day that's why i'm requesting your help.

I can use Pibooth fine with a window size Pibooth but each time i try to use Pibooth in full size there is the "Oops something wrong happens" screen before the countdown to take the picture. 

I'm using a Raspberry Pi 4 4gb with a raspberry pi camera V1 and a Canon EOS 700D with le latest version of Pibooth.

Here is my config file : 

[GENERAL]
# User interface language: 'de', 'dk', 'en', 'es', 'fr', 'hu', 'nl' or 'no'
language = fr

# Path to save pictures (list of quoted paths accepted)
directory = ~/Pictures/pibooth/test

# Start pibooth at Raspberry Pi startup
autostart = True

# In debug mode, exceptions are not caught, logs are more verbose, pictures are cleared at startup
debug = False

# Path to custom plugin(s) not installed with pip (list of quoted paths accepted)
plugins = 

# Plugin names to be disabled after startup (list of quoted names accepted)
plugins_disabled = ('pibooth_sound_effects',)

# Enable a virtual keyboard in the settings interface
vkeyboard = False

[WINDOW]
# The (width, height) of the display window or 'fullscreen'
size = (800, 480)

# Background RGB color or image path
background = (0, 0, 0)

# Text RGB color
text_color = (255, 255, 255)

# Blinking background when a capture is taken
flash = False

# Animate the last taken picture by displaying captures one by one
animate = False

# How long is displayed the capture in seconds before switching to the next one
animate_delay = 0.2

# On 'finish' state: how long is displayed the final picture in seconds (0 if never shown)
finish_picture_delay = 5

# On 'wait' state: how long is displayed the final picture in seconds before being hidden (-1 if never hidden)
wait_picture_delay = -1

# How long is displayed the 'chosen' state:  (0 if never shown)
chosen_delay = 4

# Show arrows to indicate physical buttons: 'bottom', 'top', 'hidden' or 'touchscreen'
arrows = hidden

# Apply horizontal offset to arrows position
arrows_x_offset = 0

# How long is the preview in seconds
preview_delay = 3

# Show a countdown timer during the preview
preview_countdown = True

[... 212 lines cut ...]
  File "/usr/local/lib/python3.7/dist-packages/pibooth/camera/rpi.py", line 135, in stop_preview
    self._cam.stop_preview()
  File "/usr/lib/python3/dist-packages/picamera/camera.py", line 816, in stop_preview
    self, self._camera.outputs[self.CAMERA_PREVIEW_PORT])
  File "/usr/lib/python3/dist-packages/picamera/renderers.py", line 558, in __init__
    self.renderer.inputs[0].connect(source).enable()
  File "/usr/lib/python3/dist-packages/picamera/mmalobj.py", line 1296, in connect
    return other.connect(self, **options)
  File "/usr/lib/python3/dist-packages/picamera/mmalobj.py", line 1304, in connect
    return MMALConnection(self, other, **options)
  File "/usr/lib/python3/dist-packages/picamera/mmalobj.py", line 2141, in __init__
    super(MMALConnection, self).__init__(source, target, formats)
  File "/usr/lib/python3/dist-packages/picamera/mmalobj.py", line 1986, in __init__
    self._negotiate_format(formats)
  File "/usr/lib/python3/dist-packages/picamera/mmalobj.py", line 2044, in _negotiate_format
    mmal.MMAL_EINVAL, 'failed to negotiate port format')
picamera.exc.PiCameraMMALError: failed to negotiate port format: Argument is invalid
[ ERROR   ] pibooth           : system='Linux', node='raspberrypi', release='5.10.52-v7l+', version='#1441 SMP Tue Aug 3 18:11:56 BST 2021', machine='armv7l', processor=''
 ***********************************************************************************
 *                     Oops! It seems that pibooth has crached                     *
 *     You can report an issue on https://github.com/pibooth/pibooth/issues/new    *
 *                       and post the file: /tmp/pibooth.log                       *
 ***********************************************************************************

Thank you for your time and your help

anxuae, 2021-09-05:
Hi @Yllesia , I never tried on a 4K screen, this issue may be due to your screen resolution is too high.
Can you try on an non-4k screen?

anxuae, 2021-09-05:
I tried to made a quick fix on PR https://github.com/pibooth/pibooth/pull/281 
Unfortunately, I can not test it.

Yllesia, 2021-09-07:
Hi, thanks for your quit reply.

I don't have a non 4K screen to test if the bug is only on 4K screen but i will try to find one soon.
I will also try to use your fix to see if the problem is still there

Yllesia, 2021-09-13:
I tried to use your fix but as i'm not very familiar with github and the linux environment i don't know how to install the new version of Pibooth with the fixed files. Could you help me with that please ? 

Thanks for your help

Yllesia, 2021-09-13:
Ok i think i suceed to use your fix by downloading the developing version

It seems that the problem is still there when i'm on fullscreen with the same "Oops something wrong happens" message, but in addition the pictures are now taken with the Picamera only and not with the DLSR camera is it something you wanted with your fix ?

anxuae, 2021-09-15:
I @Yllesia, 
Did you checkout the branch max-preview-resolution ? If you are not familiar with git , the best would be to try with a not 4k screen

Yllesia, 2021-09-15:
Hi @anxuae,
i used the max-preview-resolution branch for my last test and it seems that the problem is still there.
I tried to use VNC viewer to see my raspberry desktop through my laptop screen and remove the hdmi cable from my raspberry and the 4K screen and i have exactly yhe same problem. 
Everything works fine when i'm in window size, but each time i try fullscreen i have the same error problem. Pibooth does not close itself and i can take another picture if i downsize the window and take a picture again.
It seems that the preview screen doesn't work in my install in fullscreen. Maybe is it possible to have the preview screen from the DSLR directly to try to avoid the problem from the preview with the Picamera ? or is it possible to remove the preview screen and go to the photo shot directly ?

anxuae, 2021-09-16:
It seems that wrong parameters are passed to the picamera, but can not reproduce the error. Of course if you disconnect the picamera, only DSLR is used, but your experience will be poor

Yllesia, 2021-09-16:
I will try with the DSLR only to try to debug it. 
I tried yesterday to just resize the window and the problem is happening when i go over a certain size of window not only in fullscreen.
Can you give me your config file to test it ? maybe i did something wrong in the config file

anxuae, 2021-09-16:
I had a debug print in the branch `max-preview-resolution`. Can you do a `git pull` and past the log generated until the crash?

Yllesia, 2021-09-16:
Still the same problem,

here are the logs i've got from the .log file : 

[ INFO    ] pibooth           : Installed plugins: sound-effects-0.0.3
[ INFO    ] pibooth           : Starting the photo booth application on Raspberry pi 4B
[ DEBUG   ] pibooth           : Fallback to pibooth default camera management system
[ DEBUG   ] pibooth           : Found gPhoto2 cameras on ports: 'usb:001,003'
[ INFO    ] pibooth           : Configuring hybrid camera (Picamera + gPhoto2) ...
[ DEBUG   ] pibooth           : Getting option actions/viewfinder=2
[ DEBUG   ] pibooth           : Setting option imgsettings/iso=100
[ WARNING ] pibooth           : Invalid value '100' for option iso (possible choices: ['Auto']), trying to set it anyway
[ DEBUG   ] pibooth.gphoto2   : ptp_usb_getresp [usb.c:514]: PTP_OC 0x9110 receiving resp failed: PTP Device Busy (0x2019)
[ DEBUG   ] pibooth.gphoto2   : _set_config [config.c:9128]: 'ptp_canon_eos_setdevicepropvalue (params, cursub->propid, &propval, cursub->type)' failed: PTP Device Busy (0x2019)
[ DEBUG   ] pibooth.gphoto2   : gp_context_error: The property 'ISO Speed' / 0xd103 was not set (0x2019: PTP Device Busy).
[ DEBUG   ] pibooth.gphoto2   : gp_camera_set_config [gphoto2-camera.c:1115]: 'camera->functions->set_config (camera, window, context)' failed: -110
[ ERROR   ] pibooth           : Unsupported option imgsettings/iso=100 ([-110] I/O in progress), configure your DSLR manually
[ DEBUG   ] pibooth           : Setting option settings/capturetarget=Memory card
[ INFO    ] pibooth           : Connected to printer 'Canon_MG3600_series_Printer_(Copie_3)_via_VNC_de_DESKTOP-BJ2KF54'
[ DEBUG   ] pibooth           : Activate state 'wait'
[ DEBUG   ] pibooth           : Activate state 'chosen'
[ INFO    ] pibooth           : Show picture choice (1 captures selected)
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'IHDR' 16 13
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'iCCP' 41 388
[ DEBUG   ] PIL.PngImagePlugin: iCCP profile name b'ICC profile'
[ DEBUG   ] PIL.PngImagePlugin: Compression method 0
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'bKGD' 441 6
[ DEBUG   ] PIL.PngImagePlugin: b'bKGD' 441 6 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'pHYs' 459 9
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'tIME' 480 7
[ DEBUG   ] PIL.PngImagePlugin: b'tIME' 480 7 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'tEXt' 499 25
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'IDAT' 536 8192
[ DEBUG   ] pibooth           : took 0.648 seconds
[ DEBUG   ] pibooth           : Activate state 'preview'
[ INFO    ] pibooth           : Show preview before next capture
[ ERROR   ] pibooth           : failed to negotiate port format: Argument is invalid
Traceback (most recent call last):
  File "pibooth-master/pibooth/booth.py", line 402, in main_loop
    self._machine.process(events)
  File "/home/pi/Desktop/pibooth-resolution/pibooth-master/pibooth/states.py", line 64, in process
    self.set_state(new_state_name)
  File "/home/pi/Desktop/pibooth-resolution/pibooth-master/pibooth/states.py", line 91, in set_state
    hook(cfg=self.cfg, app=self.app, win=self.win)
  File "/usr/local/lib/python3.7/dist-packages/pluggy/hooks.py", line 286, in __call__
    return self._hookexec(self, self.get_hookimpls(), kwargs)
  File "/usr/local/lib/python3.7/dist-packages/pluggy/manager.py", line 93, in _hookexec
    return self._inner_hookexec(hook, methods, kwargs)
  File "/usr/local/lib/python3.7/dist-packages/pluggy/manager.py", line 337, in traced_hookexec
    return outcome.get_result()
  File "/usr/local/lib/python3.7/dist-packages/pluggy/callers.py", line 80, in get_result
    raise ex[1].with_traceback(ex[2])
  File "/usr/local/lib/python3.7/dist-packages/pluggy/callers.py", line 52, in from_call
    result = func()
  File "/usr/local/lib/python3.7/dist-packages/pluggy/manager.py", line 335, in <lambda>
    outcome = _Result.from_call(lambda: oldcall(hook, hook_impls, kwargs))
  File "/usr/local/lib/python3.7/dist-packages/pluggy/manager.py", line 87, in <lambda>
    firstresult=hook.spec.opts.get("firstresult") if hook.spec else False,
  File "/usr/local/lib/python3.7/dist-packages/pluggy/callers.py", line 208, in _multicall
    return outcome.get_result()
  File "/usr/local/lib/python3.7/dist-packages/pluggy/callers.py", line 80, in get_result
    raise ex[1].with_traceback(ex[2])
  File "/usr/local/lib/python3.7/dist-packages/pluggy/callers.py", line 187, in _multicall
    res = hook_impl.function(*args)
  File "/home/pi/Desktop/pibooth-resolution/pibooth-master/pibooth/plugins/camera_plugin.py", line 69, in state_preview_enter
    app.camera.preview(win)
  File "/home/pi/Desktop/pibooth-resolution/pibooth-master/pibooth/camera/rpi.py", line 105, in preview
    fullscreen=False, window=tuple(rect))
  File "/usr/lib/python3/dist-packages/picamera/camera.py", line 800, in start_preview
    self, self._camera.outputs[self.CAMERA_PREVIEW_PORT], **options)
  File "/usr/lib/python3/dist-packages/picamera/renderers.py", line 477, in __init__
    self.renderer.inputs[0].connect(source).enable()
  File "/usr/lib/python3/dist-packages/picamera/mmalobj.py", line 1296, in connect
    return other.connect(self, **options)
  File "/usr/lib/python3/dist-packages/picamera/mmalobj.py", line 1304, in connect
    return MMALConnection(self, other, **options)
  File "/usr/lib/python3/dist-packages/picamera/mmalobj.py", line 2141, in __init__
    super(MMALConnection, self).__init__(source, target, formats)
[... cut ...]
[ ERROR   ] pibooth           : system='Linux', node='raspberrypi', release='5.10.52-v7l+', version='#1441 SMP Tue Aug 3 18:11:56 BST 2021', machine='armv7l', processor=''
 ***********************************************************************************
 *                     Oops! It seems that pibooth has crached                     *
 *     You can report an issue on https://github.com/pibooth/pibooth/issues/new    *
 *                       and post the file: /tmp/pibooth.log                       *
 ***********************************************************************************

Yllesia, 2021-09-16:
Everything is working with only the DSLR camera but the preview is quite slow.. is there a way to speed up the FPS from the preview with only the DSLR camera ? Or maybe just remove the preview ?

anxuae, 2021-09-16:
It seems that you didn't get the last code because I can't see the debug print in your log. It should start with `preview params ->` .

No the frame rate can't be improved due to slow communication with DSLR, that the reason why we created the `hybrid` camera: picamera + DSLR

Yllesia, 2021-09-17:
I will try again to pull the right code

anxuae, 2021-10-05:
Hi @Yllesia , do you have some news about this topic?

Yllesia, 2021-10-05:
Hi, 
As i can't resolve the problem with my 4K tv screen i recently ordered a 10inch touch screen for the raspberry i will try with it as soon as i receive it and i will post here to keep you in touch with the problem.

anxuae, 2021-10-14:
Hi @Yllesia , some news?

commit 5854da78 (2022-11-25): Merge pull request #281 from pibooth/max-preview-resolution

Fix #280: Limit the max preview resolution of Raspberry-Pi camera