pibooth/pibooth #282: Help - Overlays

swieprecht, 2021-09-14:
I'm trying to add an Overlay to pictures for my daughters Birthday Party. I think I've got everything sorted, and it all works, but I cant find any instructions or idea of how to add an overlay for her Elmo themed party.

I did try photobooth-regen (or something like that) in case it was that, via putty, but when I did that it give "KeyError: 'date' so I'm hoping thats not the way you do it and it's something obvious I'm missing?

anxuae, 2021-09-15:
Hi @swieprecht , 

Edit the configuration (`pibooth --config`) and add the path to your overlay in the key `[PICTURE] overlay`.

Note all the possible customisation are described in the configuration file

swieprecht, 2021-09-15:
Thank you @anxuae - so just to be clear, on putty, I cannot add the overlay through ~/.config/pibooth/pibooth.cfg as I have with all other options, I need to do it via the command above?

I can't see any mention of overlay in the config file at all, but I'll look again :)

werdeil, 2021-09-15:
In fact the command above will open ~/.config/pibooth/pibooth.cfg on an editor, so you can modify it directly if you want

swieprecht, 2021-09-15:
Thank you, so do I need to add the [PICTURE] overlay or should I see it somewhere?

Is there anything in the documents on this as I really can't see it and it would be incredibly helpful.

I have the default options of 4 or 1 picture - do I add one overlay for both, or seperate?

What are the dimensions for the overlay? The location I add it to?

If it's in the documentation please let me know where I've missed it, but I really cant see this anywhere

anxuae, 2021-09-16:
Hi @swieprecht , when you start pibooth for the first time, the configuration file is generated with all the default keys. It should be exactly like that: https://pibooth.readthedocs.io/en/latest/sources/config/config.html#default-configuration

Just before the section `[CAMERA]`, there is the key `overlays`. Whatever your overlay (a .PNG image file) is, just put the path to the file.

swieprecht, 2021-09-16:
Okay, progress...

I've found where to put it, can't believe I missed it before.
I've uploaded a .png file and referenced it... but it now says "something went wrong".
It does save the RAW image, but doesnt make the actual picture.
It's definitely a .png, I've made the dimensions 3600 by 2400 which is what the JPEG files were that were output, I've tried 300DPI and 96DPI. 
Any idea what I'm missing?
I've put it it in the same directory ~/Pictures/pibooth/overlay/elmo.png
[image]

anxuae, 2021-09-17:
I need the log of the crash to try to understand the problem. When crashed, path to the log is indicated in the console where pibooth was started

swieprecht, 2021-09-17:
Okay thank you I'll make the changes and send the error log tonight.

I'm assuming the issue isn't the path - I enter that in the same way as the location of where the images are saved? So the ~/Pictures/...? I'm confident on the file type and dimensions

closed 2021-09-26 by anxuae