pibooth/pibooth #283: Error after taking multiple pictures only

jmitchell637, 2021-09-14:
Hello, I get an error in '4 picture' mode with a Cannon EOS DSLR with no flash card installed. It takes the four pictures, and then immediately crashes.  The error does not occur when only taking 1 picture. 1 Picture mode works perfectly.  It seems to me like it's trying to find the internal storage of the camera, but I could be mistaken.  

```
[ INFO    ] pibooth           : +< Saving raw captures
[ DEBUG   ] pibooth           : |  gp path: /
[ DEBUG   ] pibooth.gphoto2   : |  gp_context_error: You need to specify a folder starting with /store_xxxxxxxxx/
[ DEBUG   ] pibooth.gphoto2   : |  gp_camera_file_get [gphoto2-camera.c:1693]: 'gp_filesystem_get_file (camera->fs, folder, file, type, camera_file, context)' failed: -1
[ DEBUG   ] pibooth           : +> took 0.001 seconds
[ ERROR   ] pibooth           : [-1] Unspecified error
[ DEBUG   ] pibooth           : Back to failsafe state due to error:
Traceback (most recent call last):
  File "/home/pi/pibooth/pibooth/states.py", line 50, in process
    hook(cfg=self.cfg, app=self.app, win=self.win, events=events)
  File "/usr/local/lib/python3.7/dist-packages/pluggy/_hooks.py", line 265, in __call__
    return self._hookexec(self.name, self.get_hookimpls(), kwargs, firstresult)
  File "/usr/local/lib/python3.7/dist-packages/pluggy/_manager.py", line 80, in _hookexec
    return self._inner_hookexec(hook_name, methods, kwargs, firstresult)
  File "/usr/local/lib/python3.7/dist-packages/pluggy/_manager.py", line 327, in traced_hookexec
    return outcome.get_result()
  File "/usr/local/lib/python3.7/dist-packages/pluggy/_result.py", line 60, in get_result
    raise ex[1].with_traceback(ex[2])
  File "/usr/local/lib/python3.7/dist-packages/pluggy/_result.py", line 33, in from_call
    result = func()
  File "/usr/local/lib/python3.7/dist-packages/pluggy/_manager.py", line 324, in <lambda>
    lambda: oldcall(hook_name, hook_impls, kwargs, firstresult)
  File "/usr/local/lib/python3.7/dist-packages/pluggy/_callers.py", line 60, in _multicall
    return outcome.get_result()
  File "/usr/local/lib/python3.7/dist-packages/pluggy/_result.py", line 60, in get_result
    raise ex[1].with_traceback(ex[2])
  File "/usr/local/lib/python3.7/dist-packages/pluggy/_callers.py", line 39, in _multicall
    res = hook_impl.function(*args)
  File "/home/pi/pibooth/pibooth/plugins/picture_plugin.py", line 103, in state_processing_do
    captures = app.camera.get_captures()
  File "/home/pi/pibooth/pibooth/camera/base.py", line 120, in get_captures
    images.append(self._post_process_capture(data))
  File "/home/pi/pibooth/pibooth/camera/gphoto.py", line 154, in _post_process_capture
    camera_file = self._cam.file_get(gp_path.folder, gp_path.name, gp.GP_FILE_TYPE_NORMAL)
gphoto2.GPhoto2Error: [-1] Unspecified error
[ DEBUG   ] pibooth           : Activate state 'failsafe'
[ ERROR   ] pibooth           : system='Linux', node='raspberrypi', release='5.10.17-v7l+', version='#1414 SMP Fri Apr 30 13:20:47 BST 2021', machine='armv7l', processor=''
 ***********************************************************************************
 *                     Oops! It seems that pibooth has crached                     *
 *     You can report an issue on https://github.com/pibooth/pibooth/issues/new    *
 *                       and post the file: /tmp/pibooth.log                       *
 ***********************************************************************************
```

jmitchell637, 2021-09-14:
Here's the log after a successful 1 picture capture:

```
[ INFO    ] pibooth           : +< Saving raw captures
[ DEBUG   ] pibooth           : |  gp path: /
[ DEBUG   ] PIL.TiffImagePlugin: |  tag: Make (271) - type: string (2) Tag Location: 22 - Data Location: 122 - value: b'Canon\x00'
[ DEBUG   ] PIL.TiffImagePlugin: |  tag: Model (272) - type: string (2) Tag Location: 34 - Data Location: 128 - value: b'Canon EOS DIGITAL REBEL XS\x00'
[ DEBUG   ] PIL.TiffImagePlugin: |  tag: Orientation (274) - type: short (3) - value: b'\x01\x00'
[ DEBUG   ] PIL.TiffImagePlugin: |  tag: XResolution (282) - type: rational (5) Tag Location: 58 - Data Location: 160 - value: b'H\x00\x00\x00\x01\x00\x00\x00'
[ DEBUG   ] PIL.TiffImagePlugin: |  tag: YResolution (283) - type: rational (5) Tag Location: 70 - Data Location: 168 - value: b'H\x00\x00\x00\x01\x00\x00\x00'
[ DEBUG   ] PIL.TiffImagePlugin: |  tag: ResolutionUnit (296) - type: short (3) - value: b'\x02\x00'
[ DEBUG   ] PIL.TiffImagePlugin: |  tag: DateTime (306) - type: string (2) Tag Location: 94 - Data Location: 176 - value: b'2021:09:13 18:24:16\x00'
[ DEBUG   ] PIL.TiffImagePlugin: |  tag: YCbCrPositioning (531) - type: short (3) - value: b'\x02\x00'
[ DEBUG   ] PIL.TiffImagePlugin: |  tag: ExifIFD (34665) - type: long (4) - value: b'\xc4\x00\x00\x00'
[ DEBUG   ] pibooth           : |  chosen save directory: /home/pi/Pictures/pibooth
[ DEBUG   ] pibooth           : +> took 0.981 seconds
[ INFO    ] pibooth           : +< Creating the final picture
[ INFO    ] pibooth           : |  +< Use OpenCvPictureFactory to create background
[ DEBUG   ] pibooth           : |  +> took 0.073 seconds
[ INFO    ] pibooth           : |  +< Use OpenCvPictureFactory to concatenate images
[ DEBUG   ] pibooth           : |  +> took 0.309 seconds
[ INFO    ] pibooth           : |  +< Use OpenCvPictureFactory to assemble final image
[ DEBUG   ] pibooth           : |  +> took 0.116 seconds
[ INFO    ] pibooth           : |  +< Use OpenCvPictureFactory to draw texts
[ DEBUG   ] pibooth           : |  +> took 0.186 seconds
[ DEBUG   ] pibooth           : +> took 0.692 seconds
[ INFO    ] pibooth           : +< Save image '/home/pi/Pictures/pibooth/2021-09-14-18-25-09_pibooth.jpg'
[ DEBUG   ] pibooth           : +> took 0.251 seconds
[ DEBUG   ] pibooth           : Activate state 'print'
[ INFO    ] pibooth           : +< Display the final picture

```

anxuae, 2021-09-15:
Hi @jmitchell637 , 

Probably the RAM of your Canon is too small for 4 captures. Be sure to use a SD card in your camera

jmitchell637, 2021-09-15:
Oh interesting, I had thought each picture was transferred after it was
taken.  Thanks!  I'll try to find an SD card.

anxuae, 2021-09-22:
Of course but in case of no countdown, there is no time to download, thus better is to take time during processing

anxuae, 2021-09-22:
But you can propose a PR if you have an Idea

closed 2021-09-22 by anxuae