pibooth/pibooth #284: Differentiate UI font from picture fonts

qchatenet, 2021-09-15:
Hi,

I was wondering if there is a possibility to differentiate the font used for screen's texts such as "Take a photo" and those used in the picture template ?

Currently the first font defined in config file is used for the user interface. Indeed, some fonts are great for picture rendering but less appropriate for user interface.

Thank you in advance.

werdeil, 2021-09-15:
Yes you can, you have a font option in the config for the picture, you can even have a different fonts on each line.
https://pibooth.readthedocs.io/en/latest/sources/start.html?highlight=font#texts-and-fonts

qchatenet, 2021-09-15:
Thank you for your response. Meanwhile, I was wondering if you can set 3
different fonts :
- one for GUI interface
- one for text footer 1
- one for text footer 2

Currently when you set text footer font, the sale font is used for GUI
texts.

werdeil, 2021-09-15:
Indeed I looked again and we are using the first picture font as the window font
https://github.com/pibooth/pibooth/blob/7727a7758cc2065bbff82d16ba6b9b0b981d6784/pibooth/booth.py#L156
Maybe we can add a new setting on the config for the window font, even if I'm not a big fan of adding more and more element in the config.

@anxuae what do you think?

qchatenet, 2021-09-16:
I understand your preoccupation. My suggestion aims to make the interface
invariant in regard to picture configuration. In some cases, fonts used for
picture text footers are not suitable for interface.

anxuae, 2021-09-16:
I will include it in next release

werdeil, 2021-10-03:
Closing it as the change have been merged, it will be available in next release.

closed 2021-10-03 by werdeil