pibooth/pibooth #285: Assets

DennisD88, 2021-09-17:
Is it possible to change the assets images? So can I create a own look and feel.

anxuae, 2021-09-18:
We could generate an asset folder in the config directory, the user will be responsible to set asset with correct size.

@werdeil what your feeling?

DennisD88, 2021-09-19:
Where can I find the asset folder on the raspberry, I search for it but I can't find.

anxuae, 2021-09-19:
It should be there: ~/. local/lib/pythonX.Y/site-packages/pibooth

DennisD88, 2021-09-19:
Thank for your quick response. I look ad the lib folder. I have a few Python version;

- Python 2.7
- Python 3
- Python 3.7
No of them has a folder with the name site-packages. When I search in the lib folder on site-packages or pibooth I find no folder.

anxuae, 2021-09-19:
Try this command in a console:
```
python3 -c "import pibooth;print(pibooth.__file__)"
```

It should print the location on the pibooth directory

DennisD88, 2021-09-19:
Hi,

Thanks for the commandline. I know location: /usr/local/lib/python3.7/dist-packages/pibooth/__init__.py

But when I go to this folder, my pi says the path is not valid.
De opgegeven map '/usr/local/lib/python3.7/dist-packages/pibooth/__init__.py' is niet geldig

anxuae, 2021-09-19:
The path to the assets folder is `usr/local/lib/python3.7/dist-packages/pibooth/pictures/assets`

DennisD88, 2021-09-19:
I know, but I don't find the folder dist-packages in the python3.7 folder.

anxuae, 2021-09-19:
What is the result of the command:

`sudo ls /usr/local/lib/python3.7/dist-packages/pibooth/pictures/assets`

werdeil, 2021-09-19:
> We could generate an asset folder in the config directory, the user will be responsible to update asset with correct size.
> 
> @werdeil what your feeling?

We already had a discussion about something like that some months ago (https://github.com/pibooth/pibooth/issues/122). As we are doing some processing to the asset like changing the color to match the text color or resizing them to fit the screen I find it specific enough to not see at a configurable part.

I think it is acceptable to consider that the rare user that want to change it fork the code to do it.

anxuae, 2021-09-22:
Ok, agree with you.
@DennisD88 so best is to modify your self the pibooth source code . Did you found how to to it?

If yes, can you close the issue?

Thanks

closed 2021-09-22 by DennisD88

danamaki, 2022-03-01:
Hello. When I want to change images in /usr/local/lib/python3.7/dist-packages/pibooth/pictures/assets I don't have permissions to do it in FTP. What is the solution please. Thank you and congratulations for your work.