pibooth/pibooth #286: Error : Unsupported option actions/viewfinder=0 ([-7] I/O problem) when trying to take pictures

sbnsec, 2021-09-18:
Hello, 

I get an error when trying to get pictures, after the countdown, the app immediately crash. 
I'm using a Raspberry Pi4/4Gb with a Canon PowerShot G7 X Mark 2 which is listed as compatible by g2photo. 
```
| Canon PowerShot G7 X | Image Capture, Liveview, Configuration |
```
The camera has been upgraded to the last firmware version. I reseted the configuration and the camera is on full auto mode.

Here the configuration of the application and the diagnostics file generated after the crash :
[diagnostic.log](https://github.com/pibooth/pibooth/files/7189989/diagnostic.log)
[pibooth.txt](https://github.com/pibooth/pibooth/files/7189991/pibooth.txt)

The following message is displayed : 
```sh
pi@raspberrypi:~ $ pibooth
pygame 2.0.1 (SDL 2.0.9, Python 3.7.3)
Hello from the pygame community. https://www.pygame.org/contribute.html
pygame-menu 4.0.7
[ INFO    ] pibooth           : Installed plugins: 
[ INFO    ] pibooth           : Starting the photo booth application on Raspberry pi 400
[ INFO    ] pibooth           : Configuring gPhoto2 camera ...
[ WARNING ] pibooth           : No printer configured in CUPS (see http://localhost:631)
[ INFO    ] pibooth           : Show picture choice (nothing selected)
[ INFO    ] pibooth           : Show picture choice (1 captures selected)
[ INFO    ] pibooth           : Show preview before next capture
[ INFO    ] pibooth           : Take a capture
[ ERROR   ] pibooth           : Unsupported option actions/viewfinder=0 ([-7] I/O problem), configure your DSLR manually
[ ERROR   ] pibooth           : [-7] I/O problem
[ ERROR   ] pibooth           : system='Linux', node='raspberrypi', release='5.10.60-v7l+', version='#1449 SMP Wed Aug 25 15:00:44 BST 2021', machine='armv7l', processor=''
 ***********************************************************************************
 *                     Oops! It seems that pibooth has crached                     *
 *     You can report an issue on https://github.com/pibooth/pibooth/issues/new    *
 *                       and post the file: /tmp/pibooth.log                       *
 ***********************************************************************************

```

Thank you for your time and your help 😄

anxuae, 2021-09-18:
Hi @sbnsec , I never experienced this issue. Can you retry  with setting camera in manual mode?

sbnsec, 2021-09-19:
@anxuae Yes in deed, I got it working by setting the camera in manual mode,

Thanks 👍

closed 2021-09-19 by sbnsec

laptor01, 2021-09-29:
Hi @sbnsec i have exactly the same issue with the G7X. Could you please explain how you solved the issue. Putting the camera in manual mode didn't worked for me.
[pibooth.txt](https://github.com/pibooth/pibooth/files/7254384/pibooth.txt)
[diagnostic.log](https://github.com/pibooth/pibooth/files/7254390/diagnostic.log)

Thanks