pibooth/pibooth #288: VIDIOC_QUERYCTL: Innapropriaite ioctl for device

tityann, 2021-09-23:
i try to use pibooth with arducam pivariety 16 MP on a raspberry 4 with the last raspi os 
the camera works, i already took a photo (driver ok)
when i run pibooth, the window is black and in the window behind the first, i have the message:
VIDIOC_QUERYCTL: Innapropriaite ioctl for device

could you please help me to make work pibooth
thanks

anxuae, 2021-09-24:
Hi @tityann 

Thanks for report. When you say that "camera works", which command did you use to test it?

pibooth use picamera library to handle the camera. I guess that arducam is not supported. But a driver could be developed

tityann, 2021-09-24:
hello
thanks for your answer.
arducam pivariety use libcamera
i test a photo with:
libcamera-still -t 5000 -o test.jpg

sorry for my english, i am french : i don't know what a PR is
thanks

anxuae, 2021-09-26:
PR is a "Pull Request" to propose a code with arducam driver implementation. Sorry but I can't do it, I have no arducam

tityann, 2021-09-27:
hello
i am not able to make code
i asked help here #290

tityann, 2021-10-03:
@anxuae 
hello, there were no answer on my ask help #290 
i will search myself a bit more
when you say , my camera need a driver implementation in photobbooth, i don't understand why you sayr driver
my camera has v4l2 support, for intitialisation, pibooth don't use it ?
thanks

anxuae, 2021-10-09:
Can you post the complete log file? I believe that error is coming from OpenCV library, but I need to be sure

anxuae, 2021-10-10:
I close this issue: error is coming from the invalid port in OpenCV camera

closed 2021-10-10 by anxuae