pibooth/pibooth #289: Forget this picture - error

JohnsonKeast, 2021-09-26:
**Describe the bug**
When photo is taken and shown you then have the option to forget this photo, if i click on "forget this photo" it sends it to the printer? and then im presented with the next screen which is take a photo or you can still print this photo if i click on reprint the photo it doesnt reprint and im taken back to the take 1 photo or 4 photos options page, this is seriously confusing unfortuantely and presents multiple issues when in use.

**Steps to reproduce the bug**
1. take a photo
2. try to click on forget this photo
3. photo is then sent to printer??
4. next page confuses user as tells them they can reprint 
5. clicking reprint doesnt reprint and takes you back to the take a photo page.

**Logs**
there are no error logs as this more a software issue than an error situation.

JohnsonKeast, 2021-09-26:
Sorry - new to GitHub, didnt mean to close the issue

anxuae, 2021-09-26:
I think you have click on the text "forget this photo". Butas indicated by the arrow (or hand), you have to click on the right half-screen to forget the picture.
The left half-screen is for printing the picture

werdeil, 2021-09-26:
Hi @JohnsonKeast
Can you tell us more about your setup (physical buttons or touchscreen)?

As @anxuae said if you are using touchscreen we only handle two actions, left or right half of the screen. We you are prompted to print or forget, you can click on the image (almost one side of the screen) or on the print part (the other side of the screen). If you click on the forget text you most likely click on the print side and not the picture one.

One solution for you is to change the text of the forget to state to press on the picture. We don't have such text as we want to work with physical button too.

JohnsonKeast, 2021-09-26:
Thanks so much for your both your responses, I use a mouse to click on the actual text this was the intuitive step as I had no idea you had to click next to the text let alone a specific side of the text, other photobooth users also followed this method. I initially thought the theme parsing error caused a problem with the clickable area and was causing my problem, but wasn't quite on track with that thinking. I understand using a mouse probably isn't ideal and buttons or touchscreen would work the best, I only had a mouse to test unfortunately :)

I originally had the arrows config set to hidden in the pibooth --config, after setting these to touchscreen I realized I was clicking in the wrong area as described by both these posted comments (thanks so much for clearing that up) 

I would love to be able to change the text when printing/reprinting and was wondering if you could point me to the .py file that contains the text that instructs the user to forget the photo/reprint the photo, obviously a function or some kind of print command is called to display this text but I haven't actually been able to find the location of this piece of code.

Is there some way I can buy you people a coffee or pizza? 

Thanks again for your feedback.

anxuae, 2021-09-27:
See https://pibooth.readthedocs.io/en/latest/sources/start.html#gui-translations to change text translation

closed 2021-10-04 by anxuae