pibooth/pibooth #290: Add support for Arducam camera (libcamera)

tityann, 2021-09-27:
hello
Could someone add the support for the arducam pivariety camera please
this camera use libcamera to work. 
she is compatible with openCV
my imx298 modele : [https://www.arducam.com/docs/cameras-for-raspberry-pi/pivariety/16mp-imx298-pivariety-motorized-focus-camera-module/](url)
installation of librairie : [https://www.arducam.com/docs/cameras-for-raspberry-pi/pivariety/how-to-access-and-program-the-pivariety-camera-using-command-line-python-opencv-and-gstreamer/](url)
thanks for the help

tityann, 2021-10-03:
hello
could someone tell what i could do to help this issue.
i can't do it alone
i check the source but i can't identify when the probblem come

anxuae, 2021-10-03:
If the Arducam works with OpenCV, you could create a plugin which give priority to OpenCV camera (and ignore Raspberrypi camera).
For instance, save this code in a file and setup it in th configuration file:

```python
"""Plugin to use OpenCV and gPhoto2 camera."""

import time
import pibooth
from pibooth.utils import LOGGER
from pibooth import camera

__version__ = "1.0.0"

@pibooth.hookimpl
def pibooth_setup_camera(cfg):
    cv_cam_proxy = camera.get_cv_camera_proxy()
    gp_cam_proxy = camera.get_gp_camera_proxy()

    if cv_cam_proxy and gp_cam_proxy:
        LOGGER.info("Configuring hybrid camera with retry (OpenCV + gPhoto2) ...")
        return camera.HybridCvCamera(cv_cam_proxy, gp_cam_proxy)
    elif gp_cam_proxy:
        LOGGER.info("Configuring gPhoto2 camera with retry ...")
        return camera.GpCamera(gp_cam_proxy)
    elif cv_cam_proxy:
        LOGGER.info("Configuring OpenCV camera ...")
        return camera.CvCamera(cv_cam_proxy)
```

anxuae, 2021-10-03:
Can you give me the result of the following command:

```
vcgencmd get_camera
```

tityann, 2021-10-04:
hello
thank you i will try it this evening
but before, could you explain me this
`For instance, save this code in a file and setup it in th configuration file:`
i just put the name of the file , i created in the configuration file ?
thanks

anxuae, 2021-10-04:
If you want to learn about plugins, please have a look to the documentation: https://pibooth.readthedocs.io/en/latest/sources/plugins/plugins.html

In the configuration, you will find the key `[GENERAL][plugins]` where you can put the path the file of the custom plugin

tityann, 2021-10-04:
hello
pi@raspberrypi:~ $ vcgencmd get_camera
supported=1 detected=0

tityann, 2021-10-04:
for the plugins thanks..but i will try tomorow...
it seems that in my several test, i broke something because opencv doen't work anymore in my configuration

anxuae, 2021-10-09:
Can you post the complete log file? I believe that error is coming from OpenCV library, but I need to be sure

tityann, 2021-10-09:
unfortunatly, this is the only message...repeating on every line
in loop

anxuae, 2021-10-10:
I missed `import libcamera` at the beginning of the plugin.
The comment is to indicate that you have to implement all the methods (using libcamera API of course) as it is done for OpenCV driver located in https://github.com/pibooth/pibooth/blob/master/pibooth/camera/opencv.py

anxuae, 2021-10-10:
I've quickly made a draft of the driver class. But I can debug it. Now it's your job ;-)

```python

class LibcamCamera(BaseCamera):

    """Libcamera camera management.
    """

    IMAGE_EFFECTS = [u'none',
                     u'blur',
                     u'contour',
                     u'detail',
                     u'edge_enhance',
                     u'edge_enhance_more',
                     u'emboss',
                     u'find_edges',
                     u'smooth',
                     u'smooth_more',
                     u'sharpen']

    def __init__(self, camera_proxy):
        super(LibcamCamera, self).__init__(camera_proxy)
        self._overlay_alpha = 255

    def _specific_initialization(self):
        """Camera initialization.
        """
        self._cam.initCamera(self.resolution[0], self.resolution[1],
                             libcamera.PixelFormat.RGB888, buffercount=4,
                             rotation=self.preview_rotation)
        self._cam.set(libcamera.ExposureTime, self.preview_iso)

    def _show_overlay(self, text, alpha):
        """Add an image as an overlay.
        """
        if self._window:  # No window means no preview displayed
            rect = self.get_rect()
            self._overlay_alpha = alpha
            self._overlay = self.build_overlay((rect.width, rect.height), str(text), alpha)

    def _get_preview_image(self):
        """Capture a new preview image.
        """
        rect = self.get_rect()

        ret, data = self._cam.readFrame()
        if not ret:
            raise IOError("Can not get camera preview image")

        image = Image.open(io.BytesIO(data.imageData))
        self._cam.returnFrameBuffer(data)
        # Crop to keep aspect ratio of the resolution
        image = image.crop(sizing.new_size_by_croping_ratio(image.size, self.resolution))
        # Resize to fit the available space in the window
        image = image.resize(sizing.new_size_keep_aspect_ratio(image.size, (rect.width, rect.height), 'outer'))

        if self.preview_flip:
            image = image.transpose(Image.FLIP_LEFT_RIGHT)

        if self._overlay:
            image.paste(self._overlay, (0, 0), self._overlay)
        return image

    def _post_process_capture(self, capture_data):
        """Rework capture data.

        :param capture_data: couple (frame, effect)
        :type capture_data: tuple
        """
[... 80 lines cut ...]
                             rotation=self.capture_rotation)
        if self.capture_iso != self.preview_iso:
            self._cam.set(libcamera.ExposureTime, self.capture_iso)

        LOGGER.debug("Taking capture at resolution %s", self.resolution)
        ret, data = self._cam.readFrame()
        if not ret:
            raise IOError("Can not capture frame")

        if self.capture_iso != self.preview_iso:
            self._cam.set(libcamera.ExposureTime, self.preview_iso)

        self._captures.append((data.imageData, effect))
        self._cam.returnFrameBuffer(data)
        time.sleep(0.5)  # To let time to see "Smile"

        self._hide_overlay()  # If stop_preview() has not been called

    def quit(self):
        """Close the camera driver, it's definitive.
        """
        if self._cam:
            self._cam.closeCamera()

```

anxuae, 2021-10-11:
First of all, you have 2 class LibcameraCamera in the plugin, so please remove one.

Then, it seems that takin a picture has failed... but I can not help because I can not test

tityann, 2021-10-16:
so with your driver/plugin edited on 11/10/2021
and includ the ```self._cam.startCamera()```

```javascript
import time
import pibooth
import libcamera

from pibooth.utils import LOGGER
from pibooth.camera.base import BaseCamera

__version__ = "1.0.0"

class LibcameraCamera(BaseCamera):

    """Libcamera camera management.
    """

    IMAGE_EFFECTS = [u'none',
                     u'blur',
                     u'contour',
                     u'detail',
                     u'edge_enhance',
                     u'edge_enhance_more',
                     u'emboss',
                     u'find_edges',
                     u'smooth',
                     u'smooth_more',
                     u'sharpen']

    def __init__(self, camera_proxy):
        super(LibcameraCamera, self).__init__(camera_proxy)
        self._overlay_alpha = 255

    def _specific_initialization(self):
        """Camera initialization.
        """
        self._cam.initCamera(self.resolution[0], self.resolution[1],
                             libcamera.PixelFormat.RGB888, buffercount=4,
                             rotation=self.preview_rotation)
        self._cam.set(libcamera.ExposureTime, self.preview_iso)
        framerate = 30
        frame_time = 1000000 // framerate
        self._cam.set(libcamera.FrameDurationLimits, [frame_time, frame_time])

    def _show_overlay(self, text, alpha):
        """Add an image as an overlay.
        """
        if self._window:  # No window means no preview displayed
            rect = self.get_rect()
            self._overlay_alpha = alpha
            self._overlay = self.build_overlay((rect.width, rect.height), str(te                                                                                                                                                             xt), alpha)

    def _get_preview_image(self):
        """Capture a new preview image.
        """
        rect = self.get_rect()

        ret, data = self._cam.readFrame()
        if not ret:
            raise IOError("Can not get camera preview image")

        image = Image.open(io.BytesIO(data.imageData))
        self._cam.returnFrameBuffer(data)
        # Crop to keep aspect ratio of the resolution
        image = image.crop(sizing.new_size_by_croping_ratio(image.size, self.res                                                                                                                                                             olution))
        # Resize to fit the available space in the window
        image = image.resize(sizing.new_size_keep_aspect_ratio(image.size, (rect                                                                                                                                                             .width, rect.height), 'outer'))

        if self.preview_flip:
[... 210 lines cut ...]
    result = func()
  File "/usr/local/lib/python3.7/dist-packages/pluggy/_manager.py", line 324, in <lambda>
    lambda: oldcall(hook_name, hook_impls, kwargs, firstresult)
  File "/usr/local/lib/python3.7/dist-packages/pluggy/_callers.py", line 60, in _multicall
    return outcome.get_result()
  File "/usr/local/lib/python3.7/dist-packages/pluggy/_result.py", line 60, in get_result
    raise ex[1].with_traceback(ex[2])
  File "/usr/local/lib/python3.7/dist-packages/pluggy/_callers.py", line 39, in _multicall
    res = hook_impl.function(*args)
  File "/home/pi/.local/lib/python3.7/site-packages/pibooth/plugins/camera_plugin.py", line 69, in state_preview_enter
    app.camera.preview(win)
  File "/home/pi/pibooth_driver_arducam.py", line 105, in preview
    self._window.show_image(self._get_preview_image())
  File "/home/pi/pibooth_driver_arducam.py", line 61, in _get_preview_image
    raise IOError("Can not get camera preview image")
OSError: Can not get camera preview image
[ DEBUG   ] pibooth           : Activate state 'failsafe'
[ ERROR   ] pibooth           : system='Linux', node='raspberrypi', release='5.10.63-v7l+', version='#1457 SMP Tue Sep 28 11:26:14 BST 2021', machine='armv7l', processor=''
 ***********************************************************************************
 *                     Oops! It seems that pibooth has crached                     *
 *     You can report an issue on https://github.com/pibooth/pibooth/issues/new    *
 *                       and post the file: /tmp/pibooth.log                       *
 ***********************************************************************************
[ DEBUG   ] pibooth           : Activate state 'wait'
````

tityann, 2021-10-16:
i tried to put the 
```self._cam.startCamera()```
in the section ```def _specific_initialization(self):```
and now i pass thz reponse `````raise IOError("Can not get camera preview image")````
now i have
```
File "/home/pi/pibooth_driver_arducam.py", line 64, in _get_preview_image
    image = Image.open(io.BytesIO(data.imageData))
NameError: name 'Image' is not defined
````
maybe a mistake in defining ""image" ???

anxuae, 2021-10-16:
Can you ask to libcamera team how to load image with PIL lib?

anxuae, 2021-10-17:
Hi @tityann , can you try to replace `Image.open(io.BytesIO(data.imageData))`  with `Image.fromarray(data.imageData)` ?

tityann, 2021-10-17:
hello @anxuae 
now the new issue is
`````
 File "/home/pi/pibooth_driver_arducam.py", line 67, in _get_preview_image
    image = image.crop(sizing.new_size_by_croping_ratio(image.size, self.resolution))
NameError: name 'sizing' is not defined
```

I don't manage to identify which library i have to import

tityann, 2021-10-17:
for "size" it seem to depend from
```import numpy as np```
but still the same error....
i don't manage to find information for "sizing"

tityann, 2021-10-17:
now i have
````
 image = Image.open(io.BytesIO(frame))
  File "/usr/local/lib/python3.7/dist-packages/PIL/Image.py", line 3024, in open
    "cannot identify image file %r" % (filename if filename else fp)
````

tityann, 2021-10-17:
i don't know if the camera is well intitialized
becaus if i ask to take another photo it crash with
````
  self._window.show_image(self._get_preview_image())
  File "/home/pi/pibooth_driver_arducam.py", line 69, in _get_preview_image
    raise IOError("Can not get camera preview image")
````
when i told you that i have move the init of the camera earlier in the script, do i did it well ?
i put it ther
```javascript
   def __init__(self, camera_proxy):
        super(LibcameraCamera, self).__init__(camera_proxy)
        self._overlay_alpha = 255

    def _specific_initialization(self):
        """Camera initialization.
        """
        self._cam.initCamera(self.resolution[0], self.resolution[1],
                             libcamera.PixelFormat.RGB888, buffercount=4,
                             rotation=self.preview_rotation)
        self._cam.set(libcamera.ExposureTime, self.preview_iso)
        framerate = 30
        frame_time = 1000000 // framerate
        self._cam.set(libcamera.FrameDurationLimits, [frame_time, frame_time])
        self._cam.startCamera()
```

anxuae, 2021-10-17:
No, it shall be put in the `preview` function.

anxuae, 2021-10-17:
def preview(self, window, flip=True):
        """Setup the preview.
        """
        self._window = window
        self.preview_flip = flip
        self._cam.startCamera()
        self._window.show_image(self._get_preview_image())

anxuae, 2021-10-17:
Can you post your current plugin?

tityann, 2021-10-20:
ok i will try this evening.
you commented the "ExposureTime setting", but this option is only on the capture ? , the preview have no "ExposureTime setting" but the preview is also black

anxuae, 2021-10-20:
Can you save the frame in a file and attache it? I will try to load it.

I've modified the plugin to write frames in a file `frame.txt`

anxuae, 2021-10-20:
ok, modified again, the file is now called `frame.jpg`. it should work

anxuae, 2021-12-12:
Bullseye, the new version of Raspberry Pi OS, introduce libcamera and break compatibility with pibooth (see https://www.raspberrypi.com/news/raspberry-pi-os-debian-bullseye/).

It's urgent to finalize the libcamera driver for pibooth

tityann, 2022-01-12:
hello, i had not enough time to see more on my test machine
now bullseyes only use libcamera, , is there someone else who have tried ?
what must i do to help

in the last test, the video of preview was very slow

Flo4Git, 2022-01-12:
> hello, i had not enough time to see more on my test machine
> 
> now bullseyes only use libcamera, , is there someone else who have tried ?
> 
> what must i do to help
> 
> 
> 
> in the last test, the video of preview was very slow

Hello,
I have tried, and confirmed not compatible. 
You have to install back previous version Buster. 
Regards

anxuae, 2022-04-05:
Will be solved with picamera2, waiting for official release

https://github.com/raspberrypi/picamera2

klemva, 2022-05-03:
Hi guys, am i understand Richter, that Arducam will work after picamera2 will release oficial package?

I buy one today for testing.

thank you for answer:)

anxuae, 2022-05-03:
Hi @klemva , yes it should be compatible

apicole, 2022-05-04:
> Will be solved with picamera2, waiting for official release
> 
> https://github.com/raspberrypi/picamera2

I m waiting for that since I upgraded my pi to Bullseye.. :-)

yaazzz, 2022-12-02:
Hello, do we have any update on picamera2 or any tester needed?

dpdamian, 2022-12-20:
interested in the in the timeline of the fix also, would prefer not to have to return the camera.

lxne, 2022-12-23:
Hi, I have a new Pi High Quality Camera working with bullseye and libcamera. Should it work with pibooth if I would install picamera2 0.3.7 Beta Release 6?

MikkeBoomBoom, 2023-04-23:
Hi, I’m struggling with a V3 picam as well, promised my sister in law a new system in two weeks *sigh* do you have an ETA? 😅

LoKa67, 2023-05-16:
Hi, 
I'm really interested in the release you're working on, any update regarding the release date?
Thanks

loranger, 2023-07-07:
Hello, 
I would be really happy to use my brand new pi camera v3 (and its autofocus) on my pi booth too.
Do we have any ETA ?

Thanks 👍🏻

MartinN3, 2023-08-31:
Until it's done i recommend to inform about this in README.

Hylobat, 2023-09-26:
The picam V3's auto focus  is the only thing missing to Pibooth to make it the ultimate photobooth solution. Hope it will come soon. 🤞

werdeil, 2026-09-12:
Closed: picamera2 (libcamera) support is available since pibooth 2.1.0 on Raspberry Pi OS Bookworm, see the installation page. The 3.x branch will use the same driver (see the pinned issue).

PR #293 (2023-04-13): Fix #290: Add implementation of libcamera driver
Close #290
commit b546001c (2023-04-13): Fix #290: Add implementation of libcamera driver (#293)

* Add implementation of libcamera driver

* Change camera color mode

* Fix libcamera imports

* Start integrate picamera2

* Fix some import

* Remove cv2 analysis from pylint

* Fix several pylint errors

* Fix several pylint errors

* Wording

* Fix lint in test_config

* Update libcamera driver

* Hide overlay before stopping preview

* Fix in libcamera

* Stop preview in processing state

* Move IMAGE_EFFECTS list in base camera

* colorize_pil_image() accept image without alpha

* Ensure that secene is set before update

* Force sc [... cut]
closed 2026-09-12 by werdeil