pibooth/pibooth #297: Raw mode (no image processing)

fschrempf, 2021-11-16:
First of all, thanks for the great project! Keep up the good work!

**Is your feature request related to a problem? Please describe.**
I used pibooth for a wedding without any printer or special layouts. Only to have a user interface for taking single images and showing the results. On my RPi3 generating the final image with frame, text, etc. took several seconds (10-20 s) during which no photo could be taken and people had to wait. That was a bit annoying as these images weren't even used afterwards.

**Describe the solution you'd like**
I'd like to have a "raw mode" which only downloads the resulting image from the camera and shows it to the user, but doesn't do any further image processing.

werdeil, 2021-11-16:
Hi @fschrempf, we can think of having a mode which is not doing the processing but I think it would be quite a bad experience if you don't show the final picture, it is no longer a photobooth but just a camera with countdown. 

And I don't think that doing the processing for the display only (as you are suggesting it in another thread) will make it as faster as you want.

fschrempf, 2021-11-26:
> Hi @fschrempf, we can think of having a mode which is not doing the processing but I think it would be quite a bad experience if you don't show the final picture, it is no longer a photobooth but just a camera with countdown.

No, I still want the final image to be viewed, but only the raw image. So it would still capture, download and show the image.

> And I don't think that doing the processing for the display only (as you are suggesting it in another thread) will make it as faster as you want.

I think this depends on the GUI and templating framework that is used. Rendering a decorated image to the screen using the GPU will probably be "instant" when done correctly. Rendering a processed image to a file, including scaling, etc. in software as currently done will be quite slow.

anxuae, 2022-04-05:
This feature will not be included to standard code of `pibooth`. However you can easily achieve it using a custom plugin. I've published an exemple here: https://pibooth.readthedocs.io/en/latest/sources/plugins/examples.html#setup-a-custom-picture-factory

closed 2022-04-05 by anxuae