pibooth/pibooth #298: Allow to customize the UI by using templates

fschrempf, 2021-11-16:
**Is your feature request related to a problem? Please describe.**
There is no easy way to customize the UI beyond what is exposed in the settings.

**Describe the solution you'd like**
It would be really nice to customize the UI beyond what is currently possible with the settings. Some kind of templates for the different views would be great. This would allow to adjust all the elements to the specific needs of the user. It would also allow to have a nice onscreen representation of the final image (text, frame, etc.), even without doing any post processing (see #297).

**Describe alternatives you've considered**
Doing adjustments in the code is possible but tedious and not portable at all.

werdeil, 2021-11-16:
Hi @fschrempf, UI is already quite customizable: you can already change text content, font, color. The part that is not customizable is text and image positions as these are fixed to work with the touchscreen mechanism.

fschrempf, 2021-11-26:
Sure. What I'm imagining is probably slightly out of scope for the current project as it requires to introduce some GUI framework that supports templating. Maybe something web-based!? Maybe something that is Python-based!?

Anyway this is just an idea. Feel free to close the issue or keep it open as a reminder. I would really like to have a look how this could be done, but unfortunately I won't have the time to do so anytime soon.

anxuae, 2022-04-05:
Web bas HMI could be a good idea to have a highly customizable UI. Feel free to propose a PR