pibooth/pibooth #299: Display multi line text

AndPDale, 2021-11-16:
Hi Pibooth team,

Alongside my Photo Booth, I am running a webserver to enable users access to the photographs via their phones. I have written a plugin to display a "See instructions to view on your phone" after finish state. (Code below - I have removed my attempts at multi line fixes, to simplify and ensure working code)
However, as I am using a 10" screen I need the text to span two lines to ensure the font is big enough. Currently i cannot get it to move on to a second line.
Please advise on if you have any recommendations?

Plugin name: view_on_phone_message.py

```
"""Plugin to display 'View on your phone' at screen after 'finish' state."""

import time
import pygame
import pibooth
from pibooth import pictures, fonts

__version__ = "0.0.1"

@pibooth.hookimpl
def state_finish_exit(win):
    win_rect = win.surface.get_rect()
    text = '''See Instructions to view on your phone!'''

    # Get best font size according to window size
    font = fonts.get_pygame_font(text, fonts.CURRENT,
                                 win_rect.width//1.5, win_rect.height//1.5)

    # Build a surface to display at screen
    text_surface = font.render(text, True, win.text_color)

    # Clear screen
    if isinstance(win.bg_color, (tuple, list)):
        win.surface.fill(win.bg_color)
    else:
        bg_surface = pictures.get_pygame_image(win.bg_color, win_rect.size, crop=True, color=None)
        win.surface.blit(bg_surface, (0, 0))

    # Draw the surface at screen
    win.surface.blit(text_surface, text_surface.get_rect(center=win_rect.center).topleft)

    # Force screen update and events process
    pygame.display.update()
    pygame.event.pump()

    # Wait 3s
    time.sleep(3)
```

werdeil, 2021-11-18:
Hi @AndPDale 

We have a custom method to display text in multi line in our background class, I think you can use it in your plugin https://github.com/pibooth/pibooth/blob/41a5abb627d78813aca34e34ba2a3ff8b2825c5d/pibooth/view/background.py#L15

Another solution: instead of having a plugin to display the text have you tried to change the text of the finished state? It may fit directly your need.

P.S.: for the web server part what did you use? We documented a while back a way using lychee but I can't find the page. If your solution is easy to put in place I'll be glad to include it in the doc.

AndPDale, 2021-11-19:
Hi @werdeil,

Thank you for providing the recommendation to use the multiline_text_to_surfaces class. Please could you provide an example in regards to how to implement such a class in my plugin code above? I am struggling to understand how to convert my 'text' line to multi-line, do I need newline characters (/n)? and then how to create the required screens to display it. Any help is much appreciated. Thanks

In terms of a webserver, I am running an apache web server with Pibooth exporting the pictures to /var/www/html/photoimages directory. To enable users to view it on their phone, I am creating an access point through the Raspberry pi, so the users connect directly to the raspberry pi via it's WiFi ssid. The users then just need to enter a static ip address. 

To avoid usage issues, I replaced the WiFi ssid and password with a QR code and the same for the website page, as not all users understand that an IP address leads them to a website. So all the users does is point their camera at the two Qr codes and select the resulting prompt. This makes it easy and quick to use, especially during a wedding. I am happy to write a full article if required.

The website is built using HTML, CSS and PHP and auto refreshes every 20 seconds to display the latest image

AndPDale, 2021-12-01:
Hi @werdeil,
I have attempted to use the multiline_text_to_surfaces method you provided previously, however the text is still displaying on a single line. I have provided the code I am using below. I would be grateful if you could let me know what I am doing wrong?

`"""Plugin to display 'View on your phone' at screen after 'finish' state."""

import time
import pygame
import pibooth
from pibooth import pictures, fonts

__version__ = "0.0.1"

@pibooth.hookimpl
def state_finish_exit(win):
    win_rect = win.surface.get_rect()
    text = '''See Instructions to
              view on your phone!'''

    # Get best font size according to window size
    font = fonts.get_pygame_font(text, fonts.CURRENT,
                                 win_rect.width//1.5, win_rect.height//1.5)

def multiline_text_to_surfaces(text, color, rect, align='center'):

    # Build a surface to display at screen
    text_surface = font.render(text, True, win.text_color)

    # Clear screen
    if isinstance(win.bg_color, (tuple, list)):
        win.surface.fill(win.bg_color)
    else:
        bg_surface = pictures.get_pygame_image(win.bg_color, win_rect.size, crop=True, color=None)
        win.surface.blit(bg_surface, (0, 0))

    # Draw the surface at screen
    win.surface.blit(text_surface, text_surface.get_rect(center=win_rect.center).topleft)

    # Force screen update and events process
    pygame.display.update()
    pygame.event.pump()

    # Wait 3s
    time.sleep(3)
`

anxuae, 2021-12-10:
Hi, 

Did you solved your issue?

AndPDale, 2021-12-12:
Hi @anxuae,

I have been unable to solve this issue. I have tried multiple methods online, without success. I am quite new to pygame syntax so may have missed something. Any advise would be great, thanks

anxuae, 2021-12-12:
```python
"""Plugin to display 'View on your phone' at screen after 'finish' state."""

import pygame
import pibooth
from pibooth.view.background import multiline_text_to_surfaces

__version__ = "0.0.1"

@pibooth.hookimpl
def state_finish_enter(win):
    win_rect = win.get_rect()
    text = "See Instructions to\nview on your phone!"
    text_color = (255, 255, 255)

    # Size of the rectangle where text will be displayed
    rect = pygame.Rect(0, 0, win_rect.width/4, win_rect.height/4)
    # Move the rectangle at bottom-right of the screen
    rect.bottomright = win_rect.bottomright
    # Let a small margin with the window borders
    rect.move_ip(-30, -10)

    for text_surface, pos in multiline_text_to_surfaces(text, text_color, rect, 'top-left'):
        win.surface.blit(text_surface, pos)
```

AndPDale, 2021-12-14:
@anxuae - thanks that now displays the text on multi-line, but it is sitting over the resulting image and cannot be seen (screenshot attached). Previously, I had the text on a new screen rather than alongside the final image, please could you advise how the code could be modified to show the message on a separate screen (screenshot example attached)

Thank you so much for your continued support 

[image]
[image]

anxuae, 2021-12-14:
sorry, I've renamed the hook because I've believed that it was a mistake.

Just rename the hook to  `state_finish_exit` and add a pause `time.sleep(3)`

closed 2021-12-16 by AndPDale