pibooth/pibooth #301: is pibooth compatible with bullseye ?

Flo4Git, 2021-11-26:
Hi, 
I have just create a fresh install using Raspberry Pi Imager, it is now installing bullseye (Debian version 11).
Running it with a pi 4, and install the softwares according to the standard documentation.
(gPhoto2 not installed ; cups and opencv installed ; pibooth[printer] installed)

Launch pibooth, it opens a grey window for few seconds, then close it and stop with message : 
OSError: Neither Raspberry Pi nor GPhoto2 nor OpenCV camera detected

I have both Camera Module v2 and HQ Camera, running well with libcamera commands.

Bullseye does not contain anymore raspistill and raspicam camera commands they are replaced by libcamera ...

** all the logs from the console **
pygame 1.9.6
Hello from the pygame community. https://www.pygame.org/contribute.html
pygame-menu 4.0.7
[ INFO    ] pibooth           : Installed plugins: 
[ INFO    ] pibooth           : Starting the photo booth application on Raspberry pi 4B
[ WARN:0] global ../modules/videoio/src/cap_gstreamer.cpp (1824) handleMessage OpenCV | GStreamer warning: Embedded video playback halted; module v4l2src0 reported: Internal data stream error.
[ WARN:0] global ../modules/videoio/src/cap_gstreamer.cpp (914) open OpenCV | GStreamer warning: unable to start pipeline
[ WARN:0] global ../modules/videoio/src/cap_gstreamer.cpp (501) isPipelinePlaying OpenCV | GStreamer warning: GStreamer: pipeline have not been created
[ WARN:0] global ../modules/videoio/src/cap_v4l.cpp (893) open VIDEOIO(V4L2:/dev/video0): can't open camera by index
[ WARN:0] global ../modules/videoio/src/cap_gstreamer.cpp (1824) handleMessage OpenCV | GStreamer warning: Embedded video playback halted; module v4l2src1 reported: Device '/dev/video1' is not a capture device.
[ WARN:0] global ../modules/videoio/src/cap_gstreamer.cpp (914) open OpenCV | GStreamer warning: unable to start pipeline
[ WARN:0] global ../modules/videoio/src/cap_gstreamer.cpp (501) isPipelinePlaying OpenCV | GStreamer warning: GStreamer: pipeline have not been created
[ WARN:0] global ../modules/videoio/src/cap_v4l.cpp (893) open VIDEOIO(V4L2:/dev/video1): can't open camera by index
[ WARN:0] global ../modules/videoio/src/cap_gstreamer.cpp (1824) handleMessage OpenCV | GStreamer warning: Embedded video playback halted; module v4l2src2 reported: Cannot identify device '/dev/video2'.
[ WARN:0] global ../modules/videoio/src/cap_gstreamer.cpp (914) open OpenCV | GStreamer warning: unable to start pipeline
[ WARN:0] global ../modules/videoio/src/cap_gstreamer.cpp (501) isPipelinePlaying OpenCV | GStreamer warning: GStreamer: pipeline have not been created
[ WARN:0] global ../modules/videoio/src/cap_v4l.cpp (893) open VIDEOIO(V4L2:/dev/video2): can't open camera by index
Traceback (most recent call last):
  File "/usr/local/bin/pibooth", line 8, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.9/dist-packages/pibooth/booth.py", line 490, in main
    app = PiApplication(config, plugin_manager)
  File "/usr/local/lib/python3.9/dist-packages/pibooth/booth.py", line 131, in __init__
    self.camera = self._pm.hook.pibooth_setup_camera(cfg=self._config)
  File "/usr/local/lib/python3.9/dist-packages/pluggy/_hooks.py", line 265, in __call__
    return self._hookexec(self.name, self.get_hookimpls(), kwargs, firstresult)
  File "/usr/local/lib/python3.9/dist-packages/pluggy/_manager.py", line 80, in _hookexec
    return self._inner_hookexec(hook_name, methods, kwargs, firstresult)
  File "/usr/local/lib/python3.9/dist-packages/pluggy/_manager.py", line 327, in traced_hookexec
    return outcome.get_result()
  File "/usr/local/lib/python3.9/dist-packages/pluggy/_result.py", line 60, in get_result
    raise ex[1].with_traceback(ex[2])
  File "/usr/local/lib/python3.9/dist-packages/pluggy/_result.py", line 33, in from_call
    result = func()
  File "/usr/local/lib/python3.9/dist-packages/pluggy/_manager.py", line 324, in <lambda>
    lambda: oldcall(hook_name, hook_impls, kwargs, firstresult)
  File "/usr/local/lib/python3.9/dist-packages/pluggy/_callers.py", line 55, in _multicall
    gen.send(outcome)
  File "/usr/local/lib/python3.9/dist-packages/pibooth/plugins/camera_plugin.py", line 26, in pibooth_setup_camera
    cam = camera.find_camera()
  File "/usr/local/lib/python3.9/dist-packages/pibooth/camera/__init__.py", line 52, in find_camera
    raise EnvironmentError("Neither Raspberry Pi nor GPhoto2 nor OpenCV camera detected")
OSError: Neither Raspberry Pi nor GPhoto2 nor OpenCV camera detected

Thank you for your help.
Regards

anxuae, 2021-11-28:
Hi @Flo4Git,

For the moment, pibooth is not compatible with libcamera.
You can find a first implementation here https://github.com/pibooth/pibooth/pull/293 but I need information about python wrapper for libcamera, can find precise documentation about it

Flo4Git, 2021-11-28:
Hi @anxuae 
Thank you for this quick answer.
OK I understand.
I have installed it on the previous version Buster, and it works well.
Thanks
Regards.

closed 2021-12-10 by anxuae

CaffeinatedFunctionality, 2022-05-10:
> Hi @Flo4Git,
> 
> For the moment, pibooth is not compatible with libcamera. You can find a first implementation here #293 but I need information about python wrapper for libcamera, but I can't find precise documentation about it

So does that PR not work? I can look into it if you can run down where you are at. I need this for my arducam I just purchased