pibooth/pibooth #302: Cannot reset PRINTER max_pages

dominiquegrota, 2021-11-29:
Your App is really Awsome !!!! Thanks a lot. I have only one problem I could not resolve, and did not find anything about it.
I use a Canon Selphy CP1300, so I use the variable max_pages to hide print button when there is no more paper (the printer does not send the information to CUPS).
It works, but I cannot reset the printer counter.

**Steps to reproduce the bug**
1. print 18 pages
2. the printer logo shows 0 and the print button is hidden
3. I open and close the settings interface (long press on both buttons to open, and long press again to close)
4. The print button is still disabled
5. The only way I can use the printer again is to reset all counters, but this deletes also the pictures on the sd card :-(

NO errors in debug mode
Even after reboot, if the number of printed pictures is greater than the [PRINTER][max_pages] in the configuration file i cannot use the printer. Either i reset all counters or i have to modify the printer max_pages value each time i reach the value (18, 36, 54, ...) and restart pibooth

**Environment**
 - Raspberry PI V4
 - OS Rasbian Burster
 - Camera Rpi cam V2
 - Pibooth version 2.0.4
 - Extra light plugin
 - Sound Plugin
 - QRcode Plugin

anxuae, 2021-12-10:
Hi,

The only way for now is to reset all counters. But your pictures are not deleted, just the counters are reset. So no problem with that point...

dominiquegrota, 2021-12-10:
Thanks for your answer, I'l have a look again for the pictures that think where deleted.
I have my printer in the photo box, so I could install a paper sensor on a GPIO port, if I get it work, I will add a PR
Thanks a lot for your amazing work !!
[image]
!

anxuae, 2021-12-21:
Hi @dominiquegrota , I'm closing this issue but feel free to send a PR if you have

closed 2021-12-21 by anxuae