pibooth/pibooth #303: Oops! It seems that pibooth has crached

Xasch, 2021-12-07:
Hello this is my first attempt to get pibooth to work.

I am using a Logitech C920 unfortunately I get an error message as soon as the photo has been taken.

In the appendix to the log

I would be very happy to help

[pibooth.log](https://github.com/pibooth/pibooth/files/7670809/pibooth.log)

anxuae, 2021-12-10:
Hi, 

It seems that your camera is not compatible with OpenCV

anxuae, 2021-12-11:
@werdeil did you already try the C920 ?

Xasch, 2021-12-11:
is there a list of cameras supported by OpenCV?

anxuae, 2021-12-11:
It's seems that there is no list. But the issue can be something else. What resolution did you put in the config? Try to set it to the max resolution of the C920.

werdeil, 2021-12-11:
> @werdeil did you already try the C920 ? I remember that there are some issues if a wrong resolution is used, but I think you have fixed it long time ago.

Yes i have a c920 and it was working last time i tried. I'll try to rebuild a pibooth with the c920 to check again. 

@Xasch can you tell us on which version of raspi OS you are, it seems that last major version has issues.

Xasch, 2021-12-11:
pi@raspberrypi:~ $ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
NAME="Raspbian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
-bash: pi@raspberrypi:~: Kommando nicht gefunden.

Xasch, 2021-12-11:
pi@raspberrypi:~ $ pibooth
pygame 2.1.0 (SDL 2.0.14, Python 3.9.2)
Hello from the pygame community. https://www.pygame.org/contribute.html
pygame-menu 4.0.7
[ INFO    ] pibooth           : Installed plugins: 
[ INFO    ] pibooth           : Starting the photo booth application on Raspberry pi 2B
[ WARN:0] global ../modules/videoio/src/cap_gstreamer.cpp (961) open OpenCV | GStreamer warning: Cannot query video position: status=0, value=-1, duration=-1
[ INFO    ] pibooth           : Configuring OpenCV camera ...
[ WARN:0] global ../modules/videoio/src/cap_gstreamer.cpp (1214) setProperty OpenCV | GStreamer warning: GStreamer: unhandled property
[ WARNING ] pibooth           : No printer configured in CUPS (see http://localhost:631)
[ INFO    ] pibooth           : Show picture choice (nothing selected)
[ INFO    ] pibooth           : Show picture choice (1 captures selected)
[ INFO    ] pibooth           : Show preview before next capture
[ INFO    ] pibooth           : Take a capture
[ WARN:0] global ../modules/videoio/src/cap_gstreamer.cpp (1824) handleMessage OpenCV | GStreamer warning: Embedded video playback halted; module v4l2src0 reported: Internal data stream error.
[ WARN:0] global ../modules/videoio/src/cap_gstreamer.cpp (536) startPipeline OpenCV | GStreamer warning: unable to start pipeline
[ WARN:0] global ../modules/videoio/src/cap_gstreamer.cpp (1085) setProperty OpenCV | GStreamer warning: no pipeline
[ INFO    ] pibooth           : Saving raw captures
[ INFO    ] pibooth           : Creating the final picture
[ INFO    ] pibooth           : Use OpenCvPictureFactory to create background
[ INFO    ] pibooth           : Use OpenCvPictureFactory to concatenate images
[ INFO    ] pibooth           : Use OpenCvPictureFactory to assemble final image
[ INFO    ] pibooth           : Use OpenCvPictureFactory to draw texts
[ INFO    ] pibooth           : Save image '/home/pi/Pictures/pibooth/2021-12-11-19-13-08_pibooth.jpg'
[ WARN:0] global ../modules/videoio/src/cap_gstreamer.cpp (501) isPipelinePlaying OpenCV | GStreamer warning: GStreamer: pipeline have not been created

werdeil, 2021-12-11:
As there are some issues with bullseye can you try reinstalling on buster (previous version of the OS)? It would help us narrow down the issue.

Xasch, 2021-12-11:
Thanks to the "old OS" it now works without any problems!

closed 2021-12-12 by anxuae