pibooth/pibooth #305: LIBSDL2 error 1

XoVooth, 2021-12-16:
*Hello,
I am a beginner and I have a problem installing LIBSDL2

after validating the command sudo apt-get install libsdl2- *, and typing YES to continue, here is what the console tells me

0 updated, 64 newly installed, 0 to remove, and 0 not updated.
It is necessary to take 110 kb / 22.3 mb in the archives.
After this operation, 328 MB of additional disk space will be used.
Would you like to continue ? [Y / n] o
Err: 1 http://raspbian.raspberrypi.org/raspbian buster / main armhf libudev-dev armhf 241-7 ~ deb10u6 + rpi1
   404 Not Found [IP: 2a00: 1098: 0: 80: 1000: 75: 0: 3 80]
E: Unable to retrieve http://raspbian.raspberrypi.org/raspbian/pool/main/s/systemd/libudev-dev_241-7~deb10u6+rpi1_armhf.deb 404 Not Found [IP: 2a00: 1098: 0: 80: 1000: 75: 0: 3 80]
E: Unable to retrieve some archives, maybe you need to run apt-get update or try with --fix-missing?

Can you help me?

thank you

anxuae, 2021-12-17:
Dis you try `sudo apt-get update` as described in the install procedure?

XoVooth, 2021-12-17:
yes yes I did the upgrade and the update

werdeil, 2021-12-17:
From what I see here: http://raspbian.raspberrypi.org/raspbian/pool/main/s/systemd/ the package available is `libudev-dev_241-7~deb10u8+rpi1_armhf.deb` I don't know why you have this issue but this is out of the scope of pibooth.

anxuae, 2021-12-21:
@XoVooth did you solve your issue? Because it's not related to pibooth, I'm closing the issue

closed 2021-12-21 by anxuae

XoVooth, 2021-12-22:
Hello,
after formatting the card and reinstalling Raspian, it's good it works !!!!