pibooth/pibooth #306: FTP upload

DennisD88, 2021-12-27:
I want to use the FTP upload plugin. But I can't place it in the plugin folder. I get a Access Denied message. Is it possible to install the plugin with a sudo command or have somebody a solution for me.

werdeil, 2021-12-27:
You can put the plugin file wherever you want and simply put the path to it in the config.

DennisD88, 2021-12-27:
Okay, I must change the config file in the root of the PiBooth folder with the path to the plugin?

werdeil, 2021-12-27:
Yes change the pibooth.cfg file (by running `pibooth --config`)

You will find a line starting with: `plugins =`, just add the path to your custom plugin here:
```
plugins = /path/to/my/plugin.py
```

Xasch, 2022-01-01:
where can I find the plugin

damienhart, 2022-01-02:
https://pypi.org/search/?q=pibooth

sravel, 2022-01-14:
Hello I also develop a SFTP plugin but I don't test on last version.
If you want to try some code you cna acces here
https://github.com/sravel/pibooth-sftp
I never push plugin on pypi so can't use pip3 install
but if you clone then install plugin with
python3 -m pip install .

anxuae, 2022-04-05:
Seems to be solved.

closed 2022-04-05 by anxuae

delmeer, 2022-10-29:
How to set the remote ftp path?