pibooth/pibooth #307: Print Orientation

c1awde, 2022-01-01:
Hey all,

I love the work you guys have done here and I'm so close to finishing my photo booth project. 

I'm not as savvy as the folks are here so forgive me if my problem is a tad rudimentary. I've successfully used the pibooth-picture-template to create a (2x6) photo strip template using the 4 picture selection. It prints out nicely (horizontally) on a Canon Selphy CP1300 however, when I change the "pictures_per_page" from 1 to 2, the copies get placed side by side in the incorrect orientation (vertically). 

My question is how can I configure pibooth/template creator/printer to change the orientation so it prints both copies in the correct (horizontal) orientation.

Below are the two different orientations. 

[image]

anxuae, 2022-04-05:
Hi @c1awde ,

No possibility for the moment, the orientation is auto selected according this function:

https://github.com/pibooth/pibooth/blob/682c4fc9d7979a8425cc150e02c9305ed65e7d05/pibooth/pictures/__init__.py#L135-L161

Maybe you can propose to me which algorithm that for for 1, 2, 3 .... n copy in order to auto select the best orientation? It will permit to avoid a new key in the configuration file.

anxuae, 2022-08-04:
Use the [pibooth-picture-template](https://github.com/pibooth/pibooth-picture-template) plugin

closed 2022-08-04 by anxuae