pibooth/pibooth #309: Can't use one raw image twice on template

golere, 2022-01-08:
Hallo everyone,

first I want to thank you all for this open-source software I'm very happy to use it for my first hobby Photobooth.

I want to print one image with 3 different photos each one should be appear twice on the template.

When using the example template everything is fine, also when I edit it a bit. As soon as I duplicate the placeholders and try to use the template I get the error:

`No template for '6' captures`

Because pibooth thinks I have 6 placeholders in the xml-file - eventhouth I only have 3 different placeholders duplicated ( 1-2-3 and 1-2-3).

Is there a workaround to get 1 capture twice on one processed image?

Thank you in advance!

werdeil, 2022-01-08:
Hi @golere, as your question is about templates and not directly pibooth can you open it in the corresponding project ? https://github.com/pibooth/pibooth-picture-template

golere, 2022-01-11:
Thanks. I repostet the Issue there :)

anxuae, 2022-04-05:
Question will be answered in [pibooth-picture-template](https://github.com/pibooth/pibooth-picture-template)

closed 2022-04-05 by anxuae