pibooth/pibooth #311: Crash after capture on Mac OS

BermudaStudio, 2022-01-21:
Hi
Trying PiBooth on Mac OS Mojave, it crashes just after the capture.
Thanks for helping.

`PIL.UnidentifiedImageError: cannot identify image file <_io.BytesIO object at 0x11f0efe50>`
[DSLR diagnostic.log](https://github.com/pibooth/pibooth/files/7914139/diagnostic.log)
[debug error log](https://github.com/pibooth/pibooth/files/7914231/error220121.txt)

**Environment**
 - Mac OS Mojave 10.14.6
 - Camera DSLR Canon EOS 70D or 60D
 - Pibooth version 2.0.4 [dslr, printer]
 - CUPS / Gphoto2 / OpenCV / Libjpeg-dev / SDL2 framework

anxuae, 2022-01-26:
Hi, can you tell me which Pillow version are you using?

BermudaStudio, 2022-01-27:
Test fails with Pillow v.8.3.1 , v.8.4 and v.9.0 …

anxuae, 2022-04-05:
@BermudaStudio do you still have the issue?

can you execute the following command and post the generated photo and the `capture.log`?

```
gphoto2 --capture-image-and-download --debug --debug-logfile=capture.log
```

BermudaStudio, 2022-04-06:
Hi,
the captured file capt0000.jpg is 0 octet !

[capture.log](https://github.com/pibooth/pibooth/files/8426788/capture.log)

anxuae, 2022-04-06:
Seems to be the reason why `pibooth` crashes: the photo is empty.
But for which reason? I don't know. Your camera was correctly connected to your Mac?

I've tried on my Mac with a Nikon D90, and it's working fine.
Maybe you cans post the `capture.log` in an issue on https://github.com/gphoto/libgphoto2

anxuae, 2022-04-22:
So that means that now you have a `capt0000.jpg` with more that o cotet? 

The issue tell me that image format is wrong or data is incomplete, can you please post the new `capture.log` ?

BermudaStudio, 2022-04-22:
Yes, the picture is complete.

[capture.log](https://github.com/pibooth/pibooth/files/8542659/capture.log)

anxuae, 2022-04-22:
Can you also post the result of the diagnosic script?

BermudaStudio, 2022-04-23:
[DSLRDiagsnostic.log](https://github.com/pibooth/pibooth/files/8546397/DSLRDiagsnostic.log)

anxuae, 2022-04-25:
Hi @BermudaStudio 

Difficult to debug for me because I can not reproduce the issue. I've made an update on the master branch. Can you install `pibooth` in developing version following those instruction: https://pibooth.readthedocs.io/en/latest/sources/dev/install.html

Then re-run the `pibooth-diag` , it should generate a `diagnostic.raw` file. Can you post it here? I would like to check that pibooth is able to get data from your camera.

anxuae, 2022-08-04:
Hi @BermudaStudio 

Any new on this topic?

anxuae, 2022-08-28:
Please re-open if you still have an issue

closed 2022-08-28 by anxuae

dpeet, 2022-10-20:
@anxuae I'm not the original poster but I had the same issue.  I downloaded the latest development version of pibooth.  I ran 
`gphoto2 --capture-image-and-download --debug --debug-logfile=capture.log` 
and was able to successfully capture (and transfer) a picture)

then I ran 
`pibooth-diag`

I put all the files in a google folder since some were large
https://drive.google.com/drive/folders/1O3n9YWYbfY_zEyZXbemNPqicK37PKjGz?usp=sharing

diagnostic.log
diagnostic.raw
capture.log
img_0655.cr3
img_0655.jpg

I'm using an m1 Macbook Pro and a Canon EOS r6.

Let me know if there's any other diagnostics to run or something else that would help!