pibooth/pibooth #313: pibooth "oops. something went wrong"

Carl8210, 2022-02-04:
Hi,
Im new to github, pibooth and pi's in general.  Most i've done is so far is install retropie and then i decided to try and make a photobooth with my pi 3b, and so I downloaded pibooth.  I've gone through the setup and had pibooth take a photo, after said photo is taken pibooth states something went wrong and goes back to the main menu.
gphoto2 capture image and download works fine so i decided to click the link from pibooth and post my log file here, Hopefully someone can help.
Thanks
[pibooth.log](https://github.com/pibooth/pibooth/files/8004775/pibooth.log)
s

Carl8210, 2022-02-05:
Still no idea whats going on, but taking a closer look at the camera, it seems that as pibooth counts down and trys to take the picture, the camera seems to receive the request but not actually take a photo, which means pibooth can't retrieve a photo and sends up the error message.  I've installed another photobooth program which works with the camera using gphoto2, but i really wanted Pibooth because of the plugin options.

anxuae, 2022-02-05:
Hi @Carl8210 , put your DSLR in manual mode. ( According to the logs, you are in `P`)

Carl8210, 2022-02-06:
Hi anxuae,
Tried that, I've tried on every setting the camera has, all bring the same error.

EmFl, 2022-03-09:
Hello,
Had a similar issue and this was due to pibooth being configured with a resolution unsupported by my webcam.
Could you check the output of (adapt device path as needed)
`v4l2-ctl --list-formats-ext -d /dev/video0` 
And then ensure that you have a match in the pibooth config file on the resolution line

```
# Resolution for camera captures (preview will have same aspect ratio)
resolution = (640, 480)
```

anxuae, 2022-04-05:
Thanks @EmFl for your feedback. I've added your advice in the documentation: https://pibooth.readthedocs.io/en/latest/sources/start.html#final-picture-resolution

closed 2022-04-05 by anxuae