pibooth/pibooth #315: green screen and chromakey

tityann, 2022-02-13:
hello i found an old plugin for pibooth which is not maintained
so could you help me to make it works please
this is the orginal script from attilad

import numpy as np
import cv2
import datetime
from random import randint
import time

def GetDateTimeString():
    dt = str(datetime.datetime.now()).split(".")[0]
    clean = dt.replace(" ","_").replace(":","_")
    return clean

def GetBackground():
    bgImage = '/home/pi/photobooth/backgrounds/3' + str(randint(1,9)) + '.jpg' #I'm starting with 3 because I have several different background sequences -- the current is 31.jpg, 32.jpg, etc
    return cv2.imread(bgImage)

def GetImage(bg):
    ret, frame = cam.read()
    
    #Mask the green screen
    hsv=cv2.cvtColor(frame,cv2.COLOR_BGR2HSV)
    image_mask=cv2.inRange(hsv,np.array([40,50,50]),np.array([80,255,255]))
    bg_mask=cv2.bitwise_and(bg,bg,mask=image_mask)
    fg_mask=cv2.bitwise_and(frame,frame,mask=cv2.bitwise_not(image_mask))
    img = cv2.add(bg_mask,fg_mask)

    return img

#Setup window for full screen
cv2.namedWindow("Photobooth", cv2.WND_PROP_FULLSCREEN)          
cv2.setWindowProperty("Photobooth", cv2.WND_PROP_FULLSCREEN, cv2.cv.CV_WINDOW_FULLSCREEN)

#options for countdown timer
fontFace = cv2.FONT_HERSHEY_SIMPLEX
fontScale = 1
thickness = 4
countdownSeconds = 5
displayPhotoSeconds = 5

#Setup WebCam
width = 640
height = 480

cam = cv2.VideoCapture(0)
cam.set(3,width)
cam.set(4,height)

bg = GetBackground()
clicked = False
clickedTime = {}

while(True):
    img = GetImage(bg) #get masked image from webcam
    
    key = cv2.waitKey(1) #check for keypress
    if clicked == True : # if countdown timer started
        elapsed = datetime.datetime.now() - clickedTime
        secs = int(elapsed.total_seconds())
        if secs > countdownSeconds : # if five seconds are up, save the current image
            clicked = False
            cv2.imwrite('/home/pi/photobooth/' + GetDateTimeString() + '.jpg',img)
            cv2.imshow('Photobooth',img)
            time.sleep(displayPhotoSeconds) # show the photo for 5 seconds 
            bg = GetBackground() # get a new background
        else : # show the countdown timer
            text = str(5-secs) + "..."
            textSize, base = cv2.getTextSize(text, fontFace, fontScale, thickness) 
            textWidth = (width - textSize[0])/2
            textHeight = (height + textSize[1])/2
            cv2.putText(img,text,(textWidth,textHeight),fontFace,fontScale,(255,255,255),thickness)
    elif key == 32 : # on spacebar pressed, start the countdown timer
        clickedTime = datetime.datetime.now()
        clicked = True
    elif key == 27 : # on escape, close the program
        break
    
    cv2.imshow('Photobooth',img) #display masked image

cv2.destroyAllWindows()
cam.release()

tityann, 2022-02-13:
the first problem is:
in line 30
cv2.set windowsproperty(photobooth",cv2.WIND_prop_FULLSCREEN, cv2.cv.CV_WINDOWS_FULLSCREEN)
attributeError : module 'cv2' has no attribute 'cv'

after modifing cv2.cv.CV_WINDOWS_FULLSCREEN in cv2.WINDOWS_FULLSCREEN , it seem to be ok..until something else

now ;

tityann, 2022-02-14:
i have read that it can be because resolution in different
son according betwwen pibooth.cfg  resolution and the sciptp i modified:
```
#Setup WebCam
width = 640
height = 480
```
into
```
#Setup WebCam
width = 2901
height = 1934
```

the result was different but still not working 
```
pygame 2.1.2 (SDL 2.0.9, Python 3.7.3)
Hello from the pygame community. https://www.pygame.org/contribute.html
pygame-menu 4.0.7
VIDIOC_STREAMON: Opération non permise
OpenCV Error: Assertion failed ((scn == 3 || scn == 4) && (depth == CV_8U || depth == CV_32F)) in cvtColor, file /build/opencv-L65chJ/opencv-3.2.0+dfsg/modules/imgproc/src/color.cpp, line 9815
Traceback (most recent call last):
  File "/usr/local/bin/pibooth", line 11, in <module>
    load_entry_point('pibooth', 'console_scripts', 'pibooth')()
  File "/home/pi/pibooth/pibooth/booth.py", line 463, in main
    config.gettuple('GENERAL', 'plugins_disabled', str))
  File "/home/pi/pibooth/pibooth/plugins/__init__.py", line 66, in load_all_plugins
    plugin = load_module(path)
  File "/home/pi/pibooth/pibooth/utils.py", line 274, in load_module
    return loader.load_module(modname)
  File "<frozen importlib._bootstrap_external>", line 407, in _check_name_wrapper
  File "<frozen importlib._bootstrap_external>", line 907, in load_module
  File "<frozen importlib._bootstrap_external>", line 732, in load_module
  File "<frozen importlib._bootstrap>", line 265, in _load_module_shim
  File "<frozen importlib._bootstrap>", line 696, in _load
  File "<frozen importlib._bootstrap>", line 677, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 728, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/home/pi/pibooth_chromakey0.py", line 52, in <module>
    img = GetImage(bg) #get masked image from webcam
  File "/home/pi/pibooth_chromakey0.py", line 20, in GetImage
    hsv=cv2.cvtColor(frame,cv2.COLOR_BGR2HSV)
cv2.error: /build/opencv-L65chJ/opencv-3.2.0+dfsg/modules/imgproc/src/color.cpp:9815: error: (-215) (scn == 3 || scn == 4) && (depth == CV_8U || depth == CV_32F) in function cvtColor
```

tityann, 2022-02-14:
@anxuae 
i need your help please

werdeil, 2022-02-14:
Hello @tityann can you tell us where you found this plugin, it may help us understand how it worked, if it worked once.

From what I see it is a standalone script (it is not using the plugins hook we have) have you tried to run it alone?

tityann, 2022-02-14:
hello, thanks for your answer
i found it here
[](https://github.com/attilad/pibooth)
i have ask to attilad about it, but i get no answer.

werdeil, 2022-02-14:
Yes it is not a plugin, as the last commit on the repo is 5 years old it is older than pibooth.

Maybe someone can make a plugin for your need but it is a quite big project. I don't think we have the free time for it.

tityann, 2022-02-17:
ok thanks you
i let the "enhancement" open, if someone knowing codding would like to add this feature to pibboth