pibooth/pibooth #318: Permissions to change pictures in assets

danamaki, 2022-03-01:
Hello. When I want to change pictures in /usr/local/lib/python3.7/dist-packages/pibooth/pictures/assets I don't have permissions to do it in FTP ( With PI acces). What is the solution please. Thank you and congratulations for your work.

werdeil, 2022-03-01:
Hello you need root access to change any file in `/usr/local/lib`
Do you have ssh access to the pi?
If yes you can upload the pictures in a folder without root access needed (for example `/home/pi`, then via ssh copy these files in the right folder (with `sudo` to give you root privileges).
```
sudo cp /home/pi/*.png /usr/local/lib/python3.7/dist-packages/pibooth/pictures/assets/
```

danamaki, 2022-03-01:
Thank you for your information. If the file already exists will it be replaced by the new one? Thank you

closed 2022-03-01 by danamaki