pibooth/pibooth #319: Support for android phone via adb

EmFl, 2022-03-09:
Hello,
I'm wondering if we could leverage old smartphones to take pictures. The webcams I have access to have very bad resolution whereas my old android phones laying around would be excellent for taking pictures.
It seems that with adb we can trigger the camera and take a picture, before I dive into this more, I'm wondering if someone already looked at this and if this could be doable or if I'm just overlooking something here.
Thanks !

werdeil, 2022-03-20:
Hi @EmFl, as far as I know, no one explored this idea.

I don't know if it is easy to have a live preview through adb, or maybe there is a way to have the android camera stream through openCV?

anxuae, 2022-04-05:
@EmFl if you want, you can propose a Pull Request to add a new camera. You need to develop a class which inherited from [BaseCamera](https://github.com/pibooth/pibooth/blob/b6a3bf6bc3f3d912800608c6df2fbf6a871184c0/pibooth/camera/base.py#L10) like is was done for [Picamera](https://github.com/pibooth/pibooth/blob/b6a3bf6bc3f3d912800608c6df2fbf6a871184c0/pibooth/camera/rpi.py#L37)

You can have some examples of how to take a capture using `add` here: https://gist.github.com/me7/12ad7c823a1b5d1a061dceb4fcd72b87

etieng, 2022-07-24:
I'm doing it using DroidCamX app and droidcam-cli
Bringing droidcam-cli up and running before launching pibooth is working (OpenCV) and preview is laggy.
Here are some resources to make it work : 
https://www.dev47apps.com/droidcam/linux/
https://github.com/dev47apps/droidcam/wiki/Raspberry-PI

EmFl, 2022-07-24:
I was testing with https://github.com/Genymobile/scrcpy as well which looks similar but the lag in preview defeats the purpose ... I'm not sure if it's related to the USB speed of the raspberry or if it's like this on any hardware. I'll try and test on another platform and report back.