pibooth/pibooth #320: "RAW" files are saved in a different folder

danamaki, 2022-03-15:
Hello. Is it possible that the "RAW" files are saved in a different folder than the montage? Thank you in advance

werdeil, 2022-03-20:
Hi @danamaki, it is already the case, raw files are stored in the raw folder, while final image are put directly in the folder. Do you mean put them on a complete different folder?

danamaki, 2022-03-21:
Yes in a completely different folder. thank you

werdeil, 2022-04-08:
Hi @danamaki I'm sorry but I don't really see any reason to make the app config more complex with this. Can you tell me a use case where you need this?

danamaki, 2022-04-08:
Yes, it's actually very simple. The photos taken are sent to "Dropbox" and I wanted to send only the montage photos and not the "Raw" folder. but I found a workaround in my Rclone command. Thanks again.

closed 2022-04-08 by danamaki