pibooth/pibooth #323: Erreur après changement d'une image dans le répertoire assets

kevhdweb, 2022-04-06:
Bonjour,

Merci pour votre code, c'est top !

J'ai essayé de remplacer l'image layout1.png dans le répertoire assets, avec les mêmes dimensions.
Mais j'ai désormais une erreur au lancement : "not enough values to unpack (expected 4, got 3).

Savez-vous ce que j'ai fait de travers ? 

Merci d'avance !

anxuae, 2022-04-06:
(back to English for the community)

Changing assets is not a feature of `pibooth`, thus strange errors may occurs.

I need the complete log file (path is displayed a the end of the crash) and your `layout1.png` to understand the reason

kevhdweb, 2022-04-06:
(Sorry for my english)

Link to download my `layout1.png` : https://drive.google.com/file/d/1Siib7JLAfHEL53CyDJ14P2-5Nb8h-4OK/view?usp=sharing

The log : 

`[ INFO    ] pibooth           : Installed plugins: sound-effects-0.0.3, picture-template-1.0.0, google-photo-1.1.0
[ INFO    ] pibooth           : Starting the photo booth application on Raspberry pi 3B
[ DEBUG   ] pibooth           : Fallback to pibooth default camera management system
[ INFO    ] pibooth           : Configuring Picamera camera ...
[ WARNING ] pibooth           : No printer found (pycups or pycups-notify not installed)
[ DEBUG   ] pibooth           : Activate state 'wait'
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'IHDR' 16 13
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'zTXt' 41 4039
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'iCCP' 4092 390
[ DEBUG   ] PIL.PngImagePlugin: iCCP profile name b'ICC profile'
[ DEBUG   ] PIL.PngImagePlugin: Compression method 0
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'bKGD' 4494 6
[ DEBUG   ] PIL.PngImagePlugin: b'bKGD' 4494 6 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'pHYs' 4512 9
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'tIME' 4533 7
[ DEBUG   ] PIL.PngImagePlugin: b'tIME' 4533 7 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'IDAT' 4552 8192
[ DEBUG   ] pibooth           : Activate state 'choose'
[ INFO    ] pibooth           : Show picture choice (nothing selected)
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'IHDR' 16 13
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'iCCP' 41 388
[ DEBUG   ] PIL.PngImagePlugin: iCCP profile name b'ICC profile'
[ DEBUG   ] PIL.PngImagePlugin: Compression method 0
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'bKGD' 441 6
[ DEBUG   ] PIL.PngImagePlugin: b'bKGD' 441 6 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'pHYs' 459 9
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'tIME' 480 7
[ DEBUG   ] PIL.PngImagePlugin: b'tIME' 480 7 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'tEXt' 499 25
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'IDAT' 536 8192
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'IHDR' 16 13
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'pHYs' 41 9
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'iTXt' 62 1628
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'IDAT' 1702 72848
[ DEBUG   ] pibooth           : took 0.278 seconds
[ ERROR   ] pibooth           : not enough values to unpack (expected 4, got 3)
[ DEBUG   ] pibooth           : Back to failsafe state due to error:
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/pibooth/states.py", line 91, in set_state
    hook(cfg=self.cfg, app=self.app, win=self.win)
  File "/usr/local/lib/python3.9/dist-packages/pluggy/_hooks.py", line 265, in __call__
    return self._hookexec(self.name, self.get_hookimpls(), kwargs, firstresult)
  File "/usr/local/lib/python3.9/dist-packages/pluggy/_manager.py", line 80, in _hookexec
    return self._inner_hookexec(hook_name, methods, kwargs, firstresult)
  File "/usr/local/lib/python3.9/dist-packages/pluggy/_manager.py", line 327, in traced_hookexec
    return outcome.get_result()
  File "/usr/local/lib/python3.9/dist-packages/pluggy/_result.py", line 60, in get_result
    raise ex[1].with_traceback(ex[2])
  File "/usr/local/lib/python3.9/dist-packages/pluggy/_result.py", line 33, in from_call
    result = func()
  File "/usr/local/lib/python3.9/dist-packages/pluggy/_manager.py", line 324, in <lambda>
    lambda: oldcall(hook_name, hook_impls, kwargs, firstresult)
  File "/usr/local/lib/python3.9/dist-packages/pluggy/_callers.py", line 60, in _multicall
    return outcome.get_result()
  File "/usr/local/lib/python3.9/dist-packages/pluggy/_result.py", line 60, in get_result
    raise ex[1].with_traceback(ex[2])
  File "/usr/local/lib/python3.9/dist-packages/pluggy/_callers.py", line 39, in _multicall
    res = hook_impl.function(*args)
  File "/usr/local/lib/python3.9/dist-packages/pibooth/plugins/view_plugin.py", line 92, in state_choose_enter
    win.show_choice(app.capture_choices)
  File "/usr/local/lib/python3.9/dist-packages/pibooth/view/window.py", line 252, in show_choice
    self._update_background(background.ChooseBackground(choices, self.arrow_location, self.arrow_offset))
  File "/usr/local/lib/python3.9/dist-packages/pibooth/view/window.py", line 124, in _update_background
    self._current_background.resize(self.surface)
  File "/usr/local/lib/python3.9/dist-packages/pibooth/view/background.py", line 334, in resize
    self.layout1 = pictures.get_pygame_layout_image(
  File "/usr/local/lib/python3.9/dist-packages/pibooth/pictures/__init__.py", line 121, in get_pygame_layout_image
    layout_image = get_pygame_image("layout{0}.png".format(layout_number),
  File "/usr/local/lib/python3.9/dist-packages/pibooth/pictures/__init__.py", line 92, in get_pygame_image
    pil_image = colorize_pil_image(pil_image, color, bg_color)
  File "/usr/local/lib/python3.9/dist-packages/pibooth/pictures/__init__.py", line 41, in colorize_pil_image
    _, _, _, alpha = pil_image.split()
ValueError: not enough values to unpack (expected 4, got 3)
[ DEBUG   ] pibooth           : Activate state 'failsafe'
[ ERROR   ] pibooth           : system='Linux', node='raspberrypi', release='5.10.92-v7+', version='#1514 SMP Mon Jan 17 17:36:39 GMT 2022', machine='armv7l', processor=''
 ***********************************************************************************
 *                     Oops! It seems that pibooth has crached                     *
 *     You can report an issue on https://github.com/pibooth/pibooth/issues/new    *
 *                       and post the file: /tmp/pibooth.log                       *
 ***********************************************************************************
[ DEBUG   ] pibooth           : Activate state 'wait'`

chonchon033, 2022-04-06:
good evening,
 I am interested in your publication, because I cannot modify this file, I do not have permission.
Thank you

kevhdweb, 2022-04-06:
> good evening,
> I am interested in your publication, because I cannot modify this file, I do not have permission.
> Thank you

[image]

anxuae, 2022-04-06:
Your image does not have alpha channel (transparency), so it is not accepted.
You can add the alpha channel by running this command in a terminal:

```bash
convert layout1.png -alpha on layout1.png
```

kevhdweb, 2022-04-06:
Thank you very much !

closed 2022-04-06 by kevhdweb

chonchon033, 2022-04-07:
Hello
I come back to you, when I copy/paste my layout1.png, it tells me:
"layout1.png: Error opening file "/user/local/lib/python3.9/dist-packages/pibooth/pictures/assets/layouts1.png": Permission denied"
If you can explain to me? Surely I'm not doing the right thing.
Thanks very much

werdeil, 2022-04-07:
@chonchon033 It seems that the file you edited doesn't have the right access on it. You need to check that the file has the same owner and rights than the previous file (or the other one in the folder).