pibooth/pibooth #324: supprimer l'impression

chonchon033, 2022-04-06:
Je voulais savoir si c'était possible de supprimer "impression de la photo" et "ou sinon tu peux toujours imprimer cette photo" de l'ecran, car je ne veux pas imprimé les photos.
Merci

anxuae, 2022-04-06:
(back to English for the community)

You can disable printer if the cups library is not installed, or if the name of the printer in the config does not correspond to a real one (for instance you can let the configuration key empty)

Details in the documentation here: https://pibooth.readthedocs.io/en/latest/sources/start.html#printer

chonchon033, 2022-04-06:
Thank you for your answer, 
but I still have the display for printing?
Thank you

anxuae, 2022-04-06:
I made a mistake in my previous comment, you have to set a wrong printer name. For instance:

```ini
# Name of the printer defined in CUPS (or use the 'default' one)
printer_name = no printer
```

In that case:

- on the `wait` state, the text "ou sinon tu peux toujours imprimer cette photo" will be hidden
- the `print` state will be hidden ( so no text "impression de la photo")

chonchon033, 2022-04-07:
Thank you very much for your help, as fast as efficient

werdeil, 2022-04-07:
Closing this issue then 👍

closed 2022-04-07 by werdeil