pibooth/pibooth #325: Layout problem

chonchon033, 2022-04-07:
Hello
I come back to you, when I copy/paste my layout1.png, it tells me:
"layout1.png: Error opening file "/user/local/lib/python3.9/dist-packages/pibooth/pictures/assets/layouts1.png": Permission denied"
If you can explain to me? Surely I'm not doing the right thing.
Thanks very much

anxuae, 2022-04-07:
You need the admin rights to write into `/user` directory
You can copy by command line: `sudo cp layout1.png /user/local/lib/python3.9/dist-packages/pibooth/pictures/assets/`

(adapt path to the `layout1.png`)

@werdeil should we cache the assets in the `.config/` folder where the user can easily access to it?

werdeil, 2022-04-07:
@anxuae you were faster than me :-)

I still don't think it is a good idea to add this feature as I still think that users shall not change the layout by themselves (even if we are helping them to do it right here).

anxuae, 2022-04-09:
@chonchon033 did you solved your issue?

closed 2022-04-11 by anxuae

chonchon033, 2022-04-12:
Hello
Yes Perfect
thank you so much