pibooth/pibooth #326: Auto-Start is not working

Jjbp6440, 2022-04-07:
**Describe the bug**
I activaed the Auto start directly from config file, we can see the value as True in the GUI, but when I start my Raspberry it's never auto started
I do have an file in the following folder /home/pi/.config/autostart but it's an empty one

**Steps to reproduce the bug**
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error.

**Logs**
NA

**Environment**
 - OS
 NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
 - Camera 
 EOS 550D
 - Pibooth version
pygame 1.9.6
Hello from the pygame community. https://www.pygame.org/contribute.html
2.0.0

werdeil, 2022-04-08:
So you have a pibooth.desktop file in this folder but it is empty?

Can you delete it manually and retry, maybe pibooth app doesn't have the right to edit/delete it.

Jjbp6440, 2022-04-08:
Hello @werdeil 
I delete it, it created a new one but still empty file and not working :(

werdeil, 2022-04-08:
I just retested and can't reproduce. Changing the option in the settings menu and then exiting it correctly add (or remove) the file and when it is added it is not empty 
```
$ cat ~/.config/autostart/pibooth.desktop
[Desktop Entry]
Name=pibooth
Exec=pibooth
Type=application
```

Can you try to fill the file manually?

Jjbp6440, 2022-04-08:
still not working :(
I added it in the file loacted it: /home/pi/.config/autostart but nothing
the file itself doesn't seem to have the rirght excecution type.
screenshot from an existing file in the same folder
[image]

Jjbp6440, 2022-04-08:
Si reset everything and now it's workin ^^
sorry to disturb you for nothing
thanks for you amzing work

werdeil, 2022-04-08:
Maybe the two files are concurrent, do you know what is the purpose of the first one? I don't have it on my side.

Can you try to put it elsewhere to keep only the `pibooth.desktop`?

And just to be sure: `pibooth` is working when you launch it in command line?

werdeil, 2022-04-08:
No problem, I'm closing this then. Have fun !

closed 2022-04-08 by werdeil