pibooth/pibooth #329: [Question] I wrote a plugin for dropbox upload

mhilcher, 2022-04-20:
Hi,

I wrote a plugin for Dropbox uploading the pictures.
It's orientated on the [pibooth-google-photo](https://github.com/pibooth/pibooth-google-photo) plugin (thx to Sébastien Ravel, Vincent Verdeil and Antoine Rousseaux) and the [updown.py](https://github.com/dropbox/dropbox-sdk-python/blob/master/example/updown.py) example from the [dropbox-sdk-python](https://github.com/dropbox/dropbox-sdk-python).

It works with a dropbox app, which can easily be created from the dropbox developers side.

It implements the app.previous_picture_url which can be used by pibooth-qrcode to produce a qr code for the last picture processed.

Now my issue:
I am not the best skilled python developer and I made this mainly for myself, because I want to use dropbox for my photobooth.
So I don't feel to be able to maintain this plugin, and making it an official one. It works for my needs but I don't know if it will work for other users.
But on the other hand there may be users who like to use this plugin and I don't want to deprive someone of using it.

Maybe you understand my dilemma.

I need some advice what to do with the plugin now.

anxuae, 2022-04-20:
Hi @mhilcher,

We can host your plugin inside the pibooth organization if you want.
In the future, you will be able to propose PR to change/update the code. But it also means that, as maintainers, @werdeil and I will be also able to make evolution on your code (of course involving you in evolutions proposal).

If you are agree with that, I will create a repo for your code.

mhilcher, 2022-04-21:
Hi @anxuae,

> We can host your plugin inside the pibooth organization if you want. In the future, you will be able to propose PR to change/update the code. But it also means that, as maintainers, @werdeil and I [...]

That would be great. I have no problem giving the code to the organization, it's just a small contribution from my side for this great project.
 
> Else you can create your own repo in your GitHub account, and share the package on pypi

I made a repo in my account and put the code into it, but I like the solution having the code here better.