pibooth/pibooth #332: Hello, bug

arthur5640, 2022-04-22:
Hello, I have an old version of Pibooth (I don't know where to check the version) with an HQ camera.
During a wedding, the photobooth bugged several times randomly.
I would like to update the version. How can I proceed without having to uninstall everything?
Thank you

anxuae, 2022-04-22:
Hi @arthur5640 ,

Just execute the command in the release note: https://github.com/pibooth/pibooth/releases/tag/2.0.5

If you are using a DSLR, you also may need to upgrade ligphoto2: follows step 5 of the install procedure: https://pibooth.readthedocs.io/en/latest/sources/install.html#install

arthur5640, 2022-04-23:
did the update however I have this problem (see photo) and I do not have access to the new configurations. moreover is it possible to change the font size of "take a picture
[image]

anxuae, 2022-04-25:
Hi @arthur5640

I can't see your picture.

If you open a terminal, you just have to enter the `pibooth` command. Then you can manage the autostart via the settings window.

arthur5640, 2022-04-26:
good morning. I want to put a shortcut on the desktop to launch pibooth, without going through the terminal

anxuae, 2022-04-26:
You have a shortcut available here: `~/.config/autostart/pibooth.desktop`

(this file is generated if you set the `autostart=True` in the configuration)

closed 2022-04-28 by anxuae