pibooth/pibooth #333: Webcam se désactive après la 1ère photo

chevelu03, 2022-04-26:
**Describe the bug**
J'utilise un PI 2B avec une webcam logitech HD Pro Webcam C920
L'application fonctionne pour la 1ère photo mais au moment où il génère le montage, on voit que la webcam s'éteint et au moment de prendre la 2ème image (soit 2ème photo unique, soit 2ème photo d'un groupe de 4 par ex) on a un message "Oups...."
[ ERROR   ] pibooth           : Can not get camera preview image
et la lumière bleue de la webcam ne se rallume pas, sauf à relancer pibooth

**Logs**
[ DEBUG   ] pibooth           : Add to buffer the image '1385088384'
[ DEBUG   ] pibooth           : took 4.435 seconds
[ DEBUG   ] pibooth           : Activate state 'chosen'
[ INFO    ] pibooth           : Show picture choice (1 captures selected)
[ DEBUG   ] pibooth           : took 2.047 seconds
[ DEBUG   ] pibooth           : Activate state 'preview'
[ INFO    ] pibooth           : Show preview before next capture
[ ERROR   ] pibooth           : Can not get camera preview image
[ DEBUG   ] pibooth           : Back to failsafe state due to error:
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/pibooth/states.py", line 98, in set_state
    hook(cfg=self.cfg, app=self.app, win=self.win)
  File "/usr/local/lib/python3.9/dist-packages/pluggy/_hooks.py", line 265, in __call__
    return self._hookexec(self.name, self.get_hookimpls(), kwargs, firstresult)
  File "/usr/local/lib/python3.9/dist-packages/pluggy/_manager.py", line 80, in _hookexec
    return self._inner_hookexec(hook_name, methods, kwargs, firstresult)
  File "/usr/local/lib/python3.9/dist-packages/pluggy/_manager.py", line 327, in traced_hookexec
    return outcome.get_result()
  File "/usr/local/lib/python3.9/dist-packages/pluggy/_result.py", line 60, in get_result
    raise ex[1].with_traceback(ex[2])
  File "/usr/local/lib/python3.9/dist-packages/pluggy/_result.py", line 33, in from_call
    result = func()
  File "/usr/local/lib/python3.9/dist-packages/pluggy/_manager.py", line 324, in <lambda>
    lambda: oldcall(hook_name, hook_impls, kwargs, firstresult)
  File "/usr/local/lib/python3.9/dist-packages/pluggy/_callers.py", line 60, in _multicall
    return outcome.get_result()
  File "/usr/local/lib/python3.9/dist-packages/pluggy/_result.py", line 60, in get_result
    raise ex[1].with_traceback(ex[2])
  File "/usr/local/lib/python3.9/dist-packages/pluggy/_callers.py", line 39, in _multicall
    res = hook_impl.function(*args)
  File "/usr/local/lib/python3.9/dist-packages/pibooth/plugins/camera_plugin.py", line 69, in state_preview_enter
    app.camera.preview(win)
  File "/usr/local/lib/python3.9/dist-packages/pibooth/camera/opencv.py", line 155, in preview
    self._window.show_image(self._get_preview_image())
  File "/usr/local/lib/python3.9/dist-packages/pibooth/camera/opencv.py", line 101, in _get_preview_image
    raise IOError("Can not get camera preview image")
OSError: Can not get camera preview image
[ DEBUG   ] pibooth           : took 0.012 seconds
[ DEBUG   ] pibooth           : Activate state 'failsafe'
[ ERROR   ] pibooth           : system='Linux', node='raspberrypi', release='5.15.34-v7+', version='#1545 SMP Fri Apr 22 11:27:19 BST 2022', machine='armv7l', processor=''
 ***********************************************************************************
 *                     Oops! It seems that pibooth has crached                     *
 *     You can report an issue on https://github.com/pibooth/pibooth/issues/new    *
 *                       and post the file: /tmp/pibooth.log                       *
 ***********************************************************************************
[ DEBUG   ] pibooth           : took 2.096 seconds
[ DEBUG   ] pibooth           : Activate state 'wait'

**Environment**
 - OS Raspbian Bull eyes
 - Logitech HD Pro Webcam C920
 - Pibooth version 2.0.5

chevelu03, 2022-04-26:
In english, sorry...

I use PI 2B with a webcam logitech HD Pro Webcam C920
Software works for 1st phot but when it mounts the photo, we can see webcam shutting off and for the 2nd image, we got a  "Oups...." message
[ ERROR ] pibooth : Can not get camera preview image
and the blue led of the webcam don't light on, I need to restart pibooth

anxuae, 2022-04-27:
@werdeil did you already tried the Webcam C920?

anxuae, 2022-04-30:
ok thanks for feedback. Bull eyes is not yet supported

closed 2022-04-30 by anxuae