pibooth/pibooth #334: [Question] Processing first photo with Nikon DSLR takes long time

mhilcher, 2022-04-27:
Hi,

I use my Nikon D5200 DSLR for taking pictures.
Always when pibooth is started the first photo takes about 20 seconds until being processed and the picture is shown.
From the second photo on it just takes a few seconds, about 4 or 5.

It doesn't really bother me because the first foto will always be a test photo, but maybe someone knows why this happens.

Greetings, Martin

anxuae, 2022-04-27:
Never observed that, but maybe the community yes?

mhilcher, 2022-05-02:
Seems to be an issue with gphoto2. I made a picture series for testing my flashlight directly with gphoto2 via command line and the same happens here, too.
The first photo takes a long time for capturing and downloading.
From the second photo on it's much faster.

mhilcher, 2022-05-19:
Ok, seems to be an issue with battery power.
Now that I use external power there is no difference between the first and the following photos.

closed 2022-05-19 by mhilcher