pibooth/pibooth #336: error when i lauch pibooth

Galelec08, 2022-05-01:
Hello, 
i have a problem pibooth not starting.
I have install the last version of Raspbian. 
My camera is a canon eos M
I have a screen 7"
My raspberry is a PI4.
who help me please? thanks a lot
this is the error messages : 

pi@Photobooth:~ $ pibooth
pygame 1.9.6
Hello from the pygame community. https://www.pygame.org/contribute.html
pygame-menu 4.0.7
[ INFO    ] pibooth           : Installed plugins: 
[ INFO    ] pibooth           : Starting the photo booth application on Raspberry pi 4B
[ WARN:0] global ../modules/videoio/src/cap_gstreamer.cpp (1824) handleMessage OpenCV | GStreamer warning: Embedded video playback halted; module v4l2src0 reported: Cannot identify device '/dev/video0'.
[ WARN:0] global ../modules/videoio/src/cap_gstreamer.cpp (914) open OpenCV | GStreamer warning: unable to start pipeline
[ WARN:0] global ../modules/videoio/src/cap_gstreamer.cpp (501) isPipelinePlaying OpenCV | GStreamer warning: GStreamer: pipeline have not been created
[ WARN:0] global ../modules/videoio/src/cap_v4l.cpp (893) open VIDEOIO(V4L2:/dev/video0): can't open camera by index
[ WARN:0] global ../modules/videoio/src/cap_gstreamer.cpp (1824) handleMessage OpenCV | GStreamer warning: Embedded video playback halted; module v4l2src1 reported: Cannot identify device '/dev/video1'.
[ WARN:0] global ../modules/videoio/src/cap_gstreamer.cpp (914) open OpenCV | GStreamer warning: unable to start pipeline
[ WARN:0] global ../modules/videoio/src/cap_gstreamer.cpp (501) isPipelinePlaying OpenCV | GStreamer warning: GStreamer: pipeline have not been created
[ WARN:0] global ../modules/videoio/src/cap_v4l.cpp (893) open VIDEOIO(V4L2:/dev/video1): can't open camera by index
[ WARN:0] global ../modules/videoio/src/cap_gstreamer.cpp (1824) handleMessage OpenCV | GStreamer warning: Embedded video playback halted; module v4l2src2 reported: Cannot identify device '/dev/video2'.
[ WARN:0] global ../modules/videoio/src/cap_gstreamer.cpp (914) open OpenCV | GStreamer warning: unable to start pipeline
[ WARN:0] global ../modules/videoio/src/cap_gstreamer.cpp (501) isPipelinePlaying OpenCV | GStreamer warning: GStreamer: pipeline have not been created
[ WARN:0] global ../modules/videoio/src/cap_v4l.cpp (893) open VIDEOIO(V4L2:/dev/video2): can't open camera by index
Traceback (most recent call last):
  File "/usr/local/bin/pibooth", line 8, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.9/dist-packages/pibooth/booth.py", line 490, in main
    app = PiApplication(config, plugin_manager)
  File "/usr/local/lib/python3.9/dist-packages/pibooth/booth.py", line 131, in __init__
    self.camera = self._pm.hook.pibooth_setup_camera(cfg=self._config)
  File "/usr/local/lib/python3.9/dist-packages/pluggy/_hooks.py", line 265, in __call__
    return self._hookexec(self.name, self.get_hookimpls(), kwargs, firstresult)
  File "/usr/local/lib/python3.9/dist-packages/pluggy/_manager.py", line 80, in _hookexec
    return self._inner_hookexec(hook_name, methods, kwargs, firstresult)
  File "/usr/local/lib/python3.9/dist-packages/pluggy/_manager.py", line 327, in traced_hookexec
    return outcome.get_result()
  File "/usr/local/lib/python3.9/dist-packages/pluggy/_result.py", line 60, in get_result
    raise ex[1].with_traceback(ex[2])
  File "/usr/local/lib/python3.9/dist-packages/pluggy/_result.py", line 33, in from_call
    result = func()
  File "/usr/local/lib/python3.9/dist-packages/pluggy/_manager.py", line 324, in <lambda>
    lambda: oldcall(hook_name, hook_impls, kwargs, firstresult)
  File "/usr/local/lib/python3.9/dist-packages/pluggy/_callers.py", line 55, in _multicall
    gen.send(outcome)
  File "/usr/local/lib/python3.9/dist-packages/pibooth/plugins/camera_plugin.py", line 26, in pibooth_setup_camera
    cam = camera.find_camera()
  File "/usr/local/lib/python3.9/dist-packages/pibooth/camera/__init__.py", line 52, in find_camera
    raise EnvironmentError("Neither Raspberry Pi nor GPhoto2 nor OpenCV camera detected")
OSError: Neither Raspberry Pi nor GPhoto2 nor OpenCV camera detected

anxuae, 2022-05-01:
Hi,

Can you try if your camera is correctly detected by running this command:

```
gphoto2 --auto-detect
```

Galelec08, 2022-05-02:
Bonjour angoisse,
Merci pour ton message. J avais déjà fait un gphoto2 --auto- detect et il trouvait bien mon appareil
 Je viens  de ré installer gphoto et la tout marche. 
Maintenant il faut que je configuré correctement pibooth car j ai des erreurs au moment de la prise de vue  !!

Voici le fichier et encore merci .
______________________________________
oment de la prise de vue une erreur apparait.
Je pense que maintenant il reste à bien configurer l'appli pibooth. je vais un peu creuser la dessus! 
voici ce qui s'affiche 
encore merci

Code :

pi@Photobooth:~ $ pibooth
pygame 1.9.6
Hello from the pygame community. https://www.pygame.org/contribute.html
pygame-menu 4.0.7
[ INFO    ] pibooth          : Installed plugins:
[ INFO    ] pibooth          : Starting the photo booth application on Raspberry pi 4B
[ WARN:0] global ../modules/videoio/src/cap_gstreamer.cpp (1824) handleMessage OpenCV | GStreamer warning: Embedded video playback halted; module v4l2src0 reported: Cannot identify device '/dev/video0'.
[ WARN:0] global ../modules/videoio/src/cap_gstreamer.cpp (914) open OpenCV | GStreamer warning: unable to start pipeline
[ WARN:0] global ../modules/videoio/src/cap_gstreamer.cpp (501) isPipelinePlaying OpenCV | GStreamer warning: GStreamer: pipeline have not been created
[ WARN:0] global ../modules/videoio/src/cap_v4l.cpp (893) open VIDEOIO(V4L2:/dev/video0): can't open camera by index
[ WARN:0] global ../modules/videoio/src/cap_gstreamer.cpp (1824) handleMessage OpenCV | GStreamer warning: Embedded video playback halted; module v4l2src1 reported: Cannot identify device '/dev/video1'.
[ WARN:0] global ../modules/videoio/src/cap_gstreamer.cpp (914) open OpenCV | GStreamer warning: unable to start pipeline
[ WARN:0] global ../modules/videoio/src/cap_gstreamer.cpp (501) isPipelinePlaying OpenCV | GStreamer warning: GStreamer: pipeline have not been created
[ WARN:0] global ../modules/videoio/src/cap_v4l.cpp (893) open VIDEOIO(V4L2:/dev/video1): can't open camera by index
[ WARN:0] global ../modules/videoio/src/cap_gstreamer.cpp (1824) handleMessage OpenCV | GStreamer warning: Embedded video playback halted; module v4l2src2 reported: Cannot identify device '/dev/video2'.
[ WARN:0] global ../modules/videoio/src/cap_gstreamer.cpp (914) open OpenCV | GStreamer warning: unable to start pipeline
[ WARN:0] global ../modules/videoio/src/cap_gstreamer.cpp (501) isPipelinePlaying OpenCV | GStreamer warning: GStreamer: pipeline have not been created
[ WARN:0] global ../modules/videoio/src/cap_v4l.cpp (893) open VIDEOIO(V4L2:/dev/video2): can't open camera by index
[ INFO    ] pibooth          : Configuring gPhoto2 camera ...
[ WARNING ] pibooth          : The connected DSLR camera is not compatible with preview
[ ERROR  ] pibooth          : Unsupported option imgsettings/iso=100 ([-2] Bad parameters), configure your DSLR manually
[ WARNING ] pibooth          : No printer found (pycups or pycups-notify not installed)
[ INFO    ] pibooth          : Show picture choice (nothing selected)
[ INFO    ] pibooth          : Show picture choice (1 captures selected)
[ INFO    ] pibooth          : Show preview before next capture
[ INFO    ] pibooth          : Take a capture
[ ERROR  ] pibooth          : [-6] Unsupported operation
[ ERROR  ] pibooth          : system='Linux', node='Photobooth', release='5.15.32-v7l+', version='#1538 SMP Thu Mar 31 19:39:41 BST 2022', machine='armv7l', processor=''
***********************************************************************************
*                    Oops! It seems that pibooth has crached                    *
*    You can report an issue on https://github.com/pibooth/pibooth/issues/new    *

anxuae, 2022-05-02:
You can run the `pibooth-diag` command and past the generated file here

anxuae, 2022-05-04:
I don't understand why openCV is used in the logs, have you got a webcam connected?
I think that you have installed Pi OS Bullseye. Pibooth is not yet compatible with this Pi OS version. Please downgrade to Buster

closed 2022-05-04 by anxuae

Galelec08, 2022-05-04:
Thanks a lot for your answer.
I don t have time to work on raspberry . I Will test this this week. Thanks
a lot
I send the result after.